FujiNetWIFI/fujinet-firmware #74: fnc-tools: FCONFIG: MAC address

FujiMan66, 2020-01-22:
MAC address incorrectly displayed.

tschak909, 2020-01-22:
Fix in esp32/tests/multilator-rev2.ino:

```
/**
   Get Adapter config.
*/
void sio_get_adapter_config()
{
  byte mac[6];
  strcpy(adapterConfig.ssid, netConfig.ssid);

#ifdef ESP8266
  strcpy(adapterConfig.hostname, WiFi.hostname().c_str());
#else
  strcpy(adapterConfig.hostname, WiFi.getHostname());
#endif

  WiFi.macAddress(mac);

  adapterConfig.localIP[0] = WiFi.localIP()[0];
  adapterConfig.localIP[1] = WiFi.localIP()[1];
  adapterConfig.localIP[2] = WiFi.localIP()[2];
  adapterConfig.localIP[3] = WiFi.localIP()[3];

  adapterConfig.gateway[0] = WiFi.gatewayIP()[0];
  adapterConfig.gateway[1] = WiFi.gatewayIP()[1];
  adapterConfig.gateway[2] = WiFi.gatewayIP()[2];
  adapterConfig.gateway[3] = WiFi.gatewayIP()[3];

  adapterConfig.netmask[0] = WiFi.subnetMask()[0];
  adapterConfig.netmask[1] = WiFi.subnetMask()[1];
  adapterConfig.netmask[2] = WiFi.subnetMask()[2];
  adapterConfig.netmask[3] = WiFi.subnetMask()[3];

  adapterConfig.dnsIP[0] = WiFi.dnsIP()[0];
  adapterConfig.dnsIP[1] = WiFi.dnsIP()[1];
  adapterConfig.dnsIP[2] = WiFi.dnsIP()[2];
  adapterConfig.dnsIP[3] = WiFi.dnsIP()[3];

  adapterConfig.macAddress[0] = mac[0]; // _WHY_ ?!
  adapterConfig.macAddress[1] = mac[1];
  adapterConfig.macAddress[2] = mac[2];
  adapterConfig.macAddress[3] = mac[3];
  adapterConfig.macAddress[4] = mac[4];
  adapterConfig.macAddress[5] = mac[5];

  sio_to_computer(adapterConfig.rawData, sizeof(adapterConfig.rawData), false);
}

```

FujiMan66, 2020-01-23:
/**
   Get Adapter config.
*/
void sio_get_adapter_config()
{
  unsigned char mac[6];
  strcpy(adapterConfig.ssid, netConfig.ssid);

#ifdef ESP8266
  strcpy(adapterConfig.hostname, WiFi.hostname().c_str());
#else
  strcpy(adapterConfig.hostname, WiFi.getHostname());
#endif

  WiFi.macAddress(mac);

  adapterConfig.localIP[0] = WiFi.localIP()[0];
  adapterConfig.localIP[1] = WiFi.localIP()[1];
  adapterConfig.localIP[2] = WiFi.localIP()[2];
  adapterConfig.localIP[3] = WiFi.localIP()[3];

  adapterConfig.gateway[0] = WiFi.gatewayIP()[0];
  adapterConfig.gateway[1] = WiFi.gatewayIP()[1];
  adapterConfig.gateway[2] = WiFi.gatewayIP()[2];
  adapterConfig.gateway[3] = WiFi.gatewayIP()[3];

  adapterConfig.netmask[0] = WiFi.subnetMask()[0];
  adapterConfig.netmask[1] = WiFi.subnetMask()[1];
  adapterConfig.netmask[2] = WiFi.subnetMask()[2];
  adapterConfig.netmask[3] = WiFi.subnetMask()[3];

  adapterConfig.dnsIP[0] = WiFi.dnsIP()[0];
  adapterConfig.dnsIP[1] = WiFi.dnsIP()[1];
  adapterConfig.dnsIP[2] = WiFi.dnsIP()[2];
  adapterConfig.dnsIP[3] = WiFi.dnsIP()[3];

  adapterConfig.macAddress[0] = mac[0];
  adapterConfig.macAddress[1] = mac[1];
  adapterConfig.macAddress[2] = mac[2];
  adapterConfig.macAddress[3] = mac[3];
  adapterConfig.macAddress[4] = mac[4];
  adapterConfig.macAddress[5] = mac[5];

  sio_to_computer(adapterConfig.rawData, sizeof(adapterConfig.rawData), false);
}

closed 2020-01-23 by FujiMan66