FujiNetWIFI/fujinet-firmware #121: PlatformIO version: TNFS Host List error after second host directory listing

jeffpiep, 2020-02-16:
Immediately after listing a TNFS directory for the second time, the ESP32 thinks it receives the SIO command CF: 00 00 00 52 6a, which is invalid. It CF: 70 f4 ..... sio_read_hosts_slots();

tschak909, 2020-02-16:
I did discover a timing error on ESP32, where the T4 delay MUST be used, otherwise, the OS mistakes a  COMPLETE for an ACK.

-Thom

jeffpiep, 2020-02-16:
DELAY_T4 fixed part of the problem. The invalid command is not longer received. However, the following actions in CONFIG repeat the error:

1. boot and attached to Wifi
2. Pick a TNFS server
3. ESC out after listing is complete
4. Pick a TNFS server

CONFIG immediately tries to return to the TNFS server list but does not ask the ESP for a list of hosts.

tschak909, 2020-02-16:
Yes, verified, and i'm trying to figure out where and how this regression creeped in.

tschak909, 2020-02-17:
fixed.

closed 2020-02-17 by tschak909