FujiNetWIFI/fujinet-firmware #133: New Disk Image command sets ATR file mode incorrectly.

tschak909, 2020-02-26:
```
Cli: 192.168.1.7 Session: 29 : DEBUG: Decoding datagram
DEBUG: cmd=29 msgsz=21
filename: edtest.atr
flags: 771
mode: 777
open: fd=3
```

mode is incorrect and should be 3 (for read/write), while flags should be 777.

-Thom

tschak909, 2020-02-26:
from the TNFS specification:
```
The flags are:
O_RDONLY        0x0001  Open read only
O_WRONLY        0x0002  Open write only
O_RDWR          0x0003  Open read/write
O_APPEND        0x0008  Append to the file, if it exists (write only)
O_CREAT         0x0100  Create the file if it doesn't exist (write only)
O_TRUNC         0x0200  Truncate the file on open for writing
O_EXCL          0x0400  With O_CREAT, returns an error if the file exists

The modes are the same as described by CHMOD (i.e. POSIX modes). These
may be modified by the server process's umask. The mode only applies
when files are created (if the O_CREAT flag is specified)
```

jeffpiep, 2020-02-29:
in diskulator_new_disk() in config, mode was not set to R/W. Set to R/W and new ATR gets mounted as R/W upon OPTION reboot.  See pull request #139. I think this fixes the problem.

closed 2020-03-02 by tschak909