FujiNetWIFI/fujinet-firmware #134: [platformIO] tnfs_readdir should skip right over System Volume Information folder.

tschak909, 2020-02-26:
As this folder is created by Windows, and can't be side-stepped, and is useless to #FujiNet users, it should be skipped over during a tnfslib::readdir()

-Thom

FujiMan66, 2020-03-14:
Is there any other files/folders that would also need to be skipped? .AppleDouble, .DS_Store, .Trash-1000, 'Network Trash Items", etc. that Windows/Mac/Linux creates for internal use?

tschak909, 2020-03-14:
Yup, pretty much all of the above! :)

-Thom

kaelef, 2020-05-26:
This is really better handled at the server side, since the client doesn't get any useful information about these files (no flag that indicates that it's hidden, for example).  Outside of that preferred option, we'd simply have to maintain a cross-platform list of names to skip.

kaelef, 2020-07-22:
Server side code now handles this

closed 2020-07-22 by kaelef