FujiNetWIFI/fujinet-firmware #141: Disks do not eject when host slot set to 0xFF

tschak909, 2020-03-02:
Disk images are still accessible after doing an eject using FEJECT from fnc-tools.

This tool ejects the disk by setting the host slot to 0xFF, as seen in fnc-tools/feject/src/main.c:
```
  if (deviceSlots.slot[s].hostSlot!=0xFF)
    {
      // unmount disk
      disk_umount(s);

      // mark disk as unmounted in deviceSlots
      deviceSlots.slot[s].hostSlot=0xFF;
      memset(deviceSlots.slot[s].file,0,sizeof(deviceSlots.slot[s].file));

      // write the deviceSlots back to FujiNet
      disk_write();

      s+=0x31;

      print("Disk D");
      printc(&s);
      print(": ejected.\x9b");
    }
  else
    {
      print("Disk D");
      s+=0x31;
      printc(&s);
      print(": not in use.\x9b");
    }
```

However, when I do the following:
```
FMOUNT 2 6 R /games1a.atr
DIR
(shows directory)
FEJECT 2
DIR
(directory still shown, instead of error 138)

@jeffpiep What behavior has changed?

tschak909, 2020-03-02:
@jeffpiep Video here: https://www.youtube.com/watch?v=3icr29zKess

tschak909, 2020-03-02:
@jeffpiep I am guessing we need to add an SIO call to close the disk image, complementing the SIO open.

jeffpiep, 2020-03-06:
SIO command 0xE9 with the device slot number in aux1 will close the disk image and null the sioDIsk file pointer.

tschak909, 2020-03-06:
Yup I verified and forgot to close. :)

closed 2020-03-06 by tschak909