FujiNetWIFI/fujinet-firmware #151: [platform.io] Printer paper breaks

tschak909, 2020-03-07:
So basically, the printer paper stops being returned.

To reproduce:

1. Erase flash
2. Upload Filesystem & upload/monitor
3. Verify that everything starts, no SPIFFS error for paper.

```
FujiNet PlatformIO Started
mounting ATR to Disk: /autorun.atr
num_para: 5760
sectorSize: 128
num_sectors: 720
mounted.
SD Card Type: SDSC
12 devices registered.
SIO Voltage: 4095
```

4. Boot into config
5. add host for irata.online, mount it.
6. select 'comm1.atr' as R/O
7. OPTION to boot
8. LOAD"D:AMODEM63.BAS"
9. LIST"P:"
10. Use browser to fetch paper
11. Debugging information from monitor window:
```
new client
GET / HTTP/1.1
Host: 192.168.1.7
User-Agent: curl/7.55.1
Accept: */*

%PDF-1.4
1 0 obj
<</Type /Catalog /Pages 2 0 R>>
endobj
3 0 obj
<</Font <</F1 4 0 R /F2 7 0 R>>>>
endobj
4 0 obj
<</Type/Font/Subtype/TrueType/Name/F1/BaseFont/PrestigeEliteNormal/Encoding/WinAnsiEncoding/FontDescriptor 5 0 R/FirstChar 32/LastChar 252/Widths 6 0 R>>
endobj
5 0 obj
<</Type/FontDescriptor/FontName/PrestigeEliteNormal/Flags 33/ItalicAngle 0/Ascent 662/Descent -216/CapHeight 662/AvgWidth 600/MaxWidth 600/FontWeight 400/XHeight 250/StemV 60/FontBBox[ -2 -216 625 662]>>
endobj
6 0 obj
[ 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600
 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600
 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600
 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600
 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600
 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600
 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 600 ]
endobj
7 0 obj
<</Type /Font /Subtype /Type1 /BaseFont /Symbol /Encoding /WinAnsiEncoding>>
endobj
8 0 obj
<</Type /Page /Parent 2 0 R /Resources 3 0 R /MediaBox [0 0 612 792] /Contents [ 9 0 R ]>>
endobj
9 0 obj
<</Length  3802>>
stream
BT
/F1 12 Tf
18 792 Td
0 -12 Td [(10 DIM P6CODE$\(204\):GOTO 10000)]TJ
[... 546 lines cut ...]
CF: 40 53 53 01 e7
ACK!
COMPLETE!
CF: 40 57 4e 01 e6
ACK!
ACK!
c:  84  x:   7.20  y: 792.00
c:  69  x:  14.40  y: 792.00
c:  83  x:  21.60  y: 792.00
c:  84  x:  28.80  y: 792.00
c: 155  x:   0.00  y: 780.00
COMPLETE!
new client
GET / HTTP/1.1
Host: 192.168.1.7
User-Agent: curl/7.55.1
Accept: */*

[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/paper) failed
client disconnected
```

From this point, a FujiNet reset will say it can't open spiffs/paper, and any attempt to fetch paper output will result in a fail.

-Thom

jeffpiep, 2020-03-09:
I found a clue in PDF object 21. It has property <</Length 00000>>, which is not right. A similar problem happened during development due to an integer overflow, which was fixed. Maybe its the same thing. I'll start there.

jeffpiep, 2020-03-09:
Are you fetching before the print job is complete?

tschak909, 2020-03-09:
I don't think so. I am loading a BASIC program, and then LIST "P:" letting
that complete, and then fetching the paper.  This works on the 820...

-Thom

jeffpiep, 2020-03-10:
I ran this on the latest master branch (with bluetooth turned off and using default flash partition) without this error. (By the way, the BASIC program has a CTRL-O special character in line 12103 that turns on the 1027 underline mode.) I downloaded it with the EDGE browser and view in Acrobat.

But, when I build with the board_build.partitions = partitions_fujinet.csv option, I do get this error. I also notice that the printing slows down near the end; my speculation is the partition is too small, SPIFFS is taking longer to write the data perhaps trying to figure out where to put it, and finally runs out of room.

tschak909, 2020-03-10:
oooof ok.
-Thom

jeffpiep, 2020-03-10:
I removed one of the two ATR images on the SPIFFS and then the test worked. I believe it worked for the 820 printer originally because it will ignore the CTRL-O for underline. The 1027, however, begins to underline, which requires 6 extra bytes per character in the PDF file. That ballooned the filesize.

closed 2020-03-10 by jeffpiep