FujiNetWIFI/fujinet-firmware #362: Short key press detection is broken

TheMontezuma, 2020-09-16:
Short key press detection is broken.  One can easily test it even without connecting to ATARI.

mozzwald, 2020-09-19:
Current timeout for SHORTPRESS is 300ms which feels too long for a normal button press. Setting it to ~160ms feels good for me https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/hardware/keys.cpp#L11

@kaelef doing the above makes it harder to get a good double tap since you need to press and release before 160ms to register for double tap. Is it possible to change the logic and make the double tap work better? Maybe wait until TAP_MAXTIME to decide if we had single or double tap?

kaelef, 2020-09-20:
Rewrote the key handling logic to make it easier to enter a short keypress

closed 2020-09-20 by kaelef