FujiNetWIFI/fujinet-firmware #365: Reboot on the second Bluetooth start

TheMontezuma, 2020-09-22:
The first long press (A button) starts Bluetooth. The second long press stops Bluetooth and starts WIFI.
The third long press causes a system reboot:

11:39:53.592 > BUTTON_A: LONG PRESS
11:39:57.981 > ACTION: Bluetooth toggle
11:39:57.981 > abort() was called at PC 0x4008581a on core 0
11:39:57.983 >
11:39:57.983 > ELF file SHA256: 7a74d13c4df7ee67
11:39:57.983 >
11:39:57.983 > Backtrace: 0x40093f39:0x3ffbcf00 0x40094301:0x3ffbcf20 0x4008581a:0x3ffbcf40 0x40085951:0x3ffbcf70 0x4021a37d:0x3ffbcf90 0x40211083:0x3ffbd250 0x40210e02:0x3ffbd2a0 0x40097135:0x3ffbd2d0 0x4009590f:0x3ffbd300 0x400e9826:0x3ffbd320 0x400ea0c8:0x3ffbd340 0x40095965:0x3ffbd370
11:39:58.014 >
11:39:58.014 > Rebooting...

I thought that it may be related to:
https://github.com/espressif/arduino-esp32/issues/2718

but the Bluetooth handling code in the fujinet-platformio seems to already have the fix for it...

TheMontezuma, 2020-09-22:
I would propose as a temporary (or maybe as a permanent) solution to make a switch to SIO2BT mode a one way trip.
After you long press a 'A' button - you enter the SIO2BT mode which stays active until you press RESET (or power off and on Fujinet).

One more benefit (besides avoiding a crash) is that you do not take a risk to come back to Fujinet mode with less resources than required (we do not know if the Bluetooth stack will free enough RAM, etc.).

kaelef, 2020-09-22:
I should be able to fix the reboot issue.  

I don't think changing to the behavior to be "permanent" would be a good move, at least not right now. If someone accidentally switches into that mode (or does so just to test and see what SIO2BT does), they have to restart to get out of it.  On the other hand, someone who wants to use SIO2BT doesn't have a need to hit that button again, so there'd be no benefit to them. 
Most of the resources used by the BT stack are committed when you enable BT support in the firmware, so there's very little benefit in turning it off after it's been compiled in.

TheMontezuma, 2020-09-22:
I proposed a workaround, because I thought you may be tired of issues I created so far ;-)
Fixing the reboot problem is the preferable solution, of course :)

kaelef, 2020-09-22:
Thanks :)
If it ends up being harder to fix than I'm assuming, then I may well go with your suggestion!

kaelef, 2020-09-24:
I haven't had any luck reproducing this problem.  Are you seeing it on a WOVER-based FN, or WROOM?

TheMontezuma, 2020-09-24:
WROVER  (the original 1.0 hardware from @mozzwald) flashed with flasher with software from 21.9 (but the same happens with software built with platformio).
I was simply long pressing the A button, to toggle between normal mode and SIO2BT mode.
The reboot happens when you try to enter SIO2BT mode the second time and it is 100% reproducible.

kaelef, 2020-09-24:
I don't know what we're doing differently, but I was able to switch back and forth between WiFi and BT on multiple tests multiple times _usually_ without a problem. However, I did get a random crash sometimes when deleting the WiFi EventGroup, which is also what happened in the stacktrace you provided.  So I just stopped deleting the WiFi EventGroup (we just reuse it if/when WiFi is restarted).  Give today's build a try when you get a chance.  I went back and forth between WiFi and BT modes many times without a crash after the change.

TheMontezuma, 2020-09-24:
I tried your commit and the crashes are gone :)
Thanks!

closed 2020-09-25 by kaelef