FujiNetWIFI/fujinet-firmware #373: CONFIG, running on Atari, should be able to show #FujiNet firmware version

billkendrick, 2020-10-17:
At the moment, you need to point a web browser to the #FujiNet's web config page to see which version of the firmware you are running (i.e., to compare with what's listed at https://fujinet.online/download/). It would be nice if this appeared in the "#FUJINET CONFIG" screen that CONFIG tool on the Atari shows when you press `[C]` on some screens.

billkendrick, 2020-10-17:
The display on the Atari should therefore re-label the CONFIG version shown (e.g., "`Version:2.5.2020-08-25`") as something less ambiguous (e.g., "`CONFIG Version: 2.6.2020-10-16`), with the new one being "`FujiNet Version: ...`" or "`Firmware Version: ...`", for example.

tschak909, 2020-10-17:
Fixed as of this morning.
[image]

closed 2020-10-17 by tschak909

billkendrick, 2020-10-17:
Thanks @tschak909 -- It looks like the "CONFIG" version disappeared, though. Wouldn't it be handy (for debugging) to mention both?

Is it possible to run a #FujiNet with CONFIG version X built-in, but instead boot onto a separate disk that runs CONFIG version Y? (Admittedly an uncommon thing to do, I think... but perhaps useful if someone wants to try out a newer build of CONFIG, or revert to an earlier version due to a regression.)