FujiNetWIFI/fujinet-firmware #378: Allow SAM to be set to P1:, P2:, and/or P3:

billkendrick, 2020-10-29:
It could be fun to map the main `P1:` / `P:` printer device to the SAM Speech Synthesizer.  Most applications (e.g., _AtariWriter_) don't provide a way to specify which printer to talk to (`P2:`, `P3:`, etc.).

Rather than making `P4:` always be SAM, would it be reasonable to allow all four printers (not just 1-3) to be configured, and allow SAM to be assigned to any of them?

(A fun side-effect could be sending different SAM configuration instructions to the different 'printers', making it easier to have multiple voices conversing with each other.)

dillera, 2026-10-03:
This issue has had no activity for about 6 years, and there have been 5044 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera

billkendrick, 2026-10-03:
I admit I still think the idea might be useful (and multiple S.A.M.'s, one per P: device, could be interesting).  But presumably no one else thinks it's of any use, so I'm fine with the ticket being closed. :)