FujiNetWIFI/fujinet-firmware #379: Support for XMM801 printer control codes

billkendrick, 2020-10-29:
I read somewhere that the XMM801 was basically an XE-styled Atari 825, however I've tried some XMM801 control codes (as described in the manual, which can be found here: https://archive.org/details/atari-xmm801-do-tmatrix-graphics-printer-owners-manual/page/n22/mode/1up) via FujiNet while it was in 825 mode, and they don't work quite as expected.

[image]

The control codes described in the manual allow controlling column width, line spacing, character pitch (pica, elite, condense, double-width), print style (bold, double-strike, underline, superscript, subscript), as well as defining both inches and lines per page and vertical tab stops, doing housekeeping (clear print buffer, initialize printer), printing "dot-addressable" graphics (regular, and high-density), using international characters, and of course moving the print head (line feed, form feed, horizontal and vertical tab, home- and end-of-line, and backspace).

(I admit this is nostalgia-driven)

jeffpiep, 2020-11-05:
Hi Bill, are these control codes also in the 825 manual or specific to the
XMM801?

billkendrick, 2020-11-25:
Consulting page 20 of the Atari 825 manual (e.g. http://www.atarimania.com/documents/Atari_825_Printer_Operators_manual.pdf), it appears that it is not the same.  e.g., Esc Ctrl-N on the XMM801 is meant to print a single line in double-width mode (see page 21 of the XMM801 manual), but the 825 emulator in FujiNet keeps it going (which seems correct for the 825).

There seem to be a lot of 825 commands the XMM801 supports (e.g., Ctrl-O to enable underline), but also alternatives it also allows (e.g., Esc Ctrl-Y, also to enable underline).

jeffpiep, 2020-12-05:
I created an XMM801 emulator based on the Epson emulator but used the XMM801 manual for the control codes. It supports the 14 typefaces chosen for the Epson emulator (choices limited to save flash space). The emulator supports normal density (esc-K) Epson graphics. It does not work with PrintShop's xX-80 mode because that uses double density 'esc-L' graphics mode. The XMM801 command for double density (called high-density in the XMM manual) is 'esc-V' - go figure.

The owner's manual on page 16 says the Atari 825 option in AtariWriter is "most compatible." [This magazine article](https://www.atarimagazines.com/v5n8/AtariPrinters.html) erroneously states the XMM801 is "fully compatible with the old Atari 825 printer."

closed 2020-12-05 by jeffpiep