FujiNetWIFI/fujinet-firmware #380: Enhancement: Web page or CONFIG. TNFS version shown current or last access

FujiMan66, 2020-10-29:
Display current/last known accessed TNFS version on webpage or CONFIG for troubleshooting issues

tschak909, 2020-10-29:
hmmm clarification? Such as if you try to access a tnfs server with the
wrong tnfs version?

-Thom

RichStephens, 2026-10-03:
Closing this one. The firmware now reads the version a TNFS server reports when it mounts, and refuses any server older than 1.2 (`tnfs_mount()` in `lib/TNFSlib/tnfslib.cpp`), so a too-old server can't cause half-working behavior any more. The clarifying question above never got an answer, either.

If it would still help to see the server's version, or a plain "server too old" error, in the web UI or CONFIG instead of only in the debug log, please open a new issue for that.

closed 2026-10-03 by RichStephens