FujiNetWIFI/fujinet-firmware #387: After ejecting all drive slots, CONFIG remains mounted

billkendrick, 2020-11-05:
# Problem
After ejecting all drive slots, CONFIG remains mounted. This makes it impossible to boot into a drive-less configuration. Appears to be a regression (per @tschak909 on Discord).

# Steps I took
- Started with Atari powered off
- Ultimate SD cartridge inserted
- FujiNet connected (no external power, so was off)
- Powered on Atari
- Chose Atari BASIC ROM cart from the list
- FujiNet booted into CONFIG
- Ensured all Fujinet drive slots were empty
- Hit [Option] to reboot

# Expected result
- Atari boots, with no drives available, into BASIC

# Actual result
- Atari boots off of the Fujinet into CONFIG

billkendrick, 2020-11-05:
Notes:
- If I hit [Option] to reboot and remove my FujiNet quickly (note: my power switch is broken, so I cannot power it off), the Atari boots into BASIC, as expected.
- If I mount something on a drive _other_ than `D1:`, it also boots into BASIC, as expected.

billkendrick, 2020-11-05:
Also this is FujiNet version `0.5.32f7ef07`

tschak909, 2020-11-19:
Now fixed in two ways:

* CONFIG now explicitly turns off the ROMDISK when leaving program.
* There is a web admin option to turn off ROMDISK, which persists.

closed 2020-11-19 by tschak909