FujiNetWIFI/fujinet-firmware #396: Debugging/info page on web interface

billkendrick, 2020-11-16:
# Purpose
Add a debugging/info page to the web interface. (I'm suggesting this, because I think the details here would be 'too much' for most people, for the main web interface.)

It could show (and perhaps even allow modification of) things like:
- current modem settings (baud rate, telnet mode, answer mode, terminal emulation (vt100, ansi, dummy, etc.), and so on)
- each drive slot's mounted image
- session id currently used when talking to any TNFS servers
- active network connection(s) (i.e., incoming call(s))

billkendrick, 2020-12-24:
I'm getting the main web UI to show what's mounted in TNFS host slots and drive slots, so crossed that off this list.

billkendrick, 2022-08-16:
Note to self: Updates would happen (as of 8/2022) in `lib/http/httpServiceParser.cpp`, and `data/BUILD_xxx/index.html`.

dillera, 2026-10-03:
This issue was opened about 6 years ago, and the code it refers to has changed since.

- `data/BUILD_xxx/index.html` no longer exists in the codebase

The files it names have changed in 84 commit(s) since then.

Closing as no longer applicable to the current firmware. If you can still reproduce this on a current firmware build, please reopen it (or open a new issue) with the firmware version, the platform, and the steps.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera