FujiNetWIFI/fujinet-firmware #397: Do not accept incoming connections if one is already active

billkendrick, 2020-11-16:
# Purpose
Do not allow incoming connections if one is already active (outgoing via ATDT, or incoming answered "call").  If we wish to keep the existing behavior, at least provide this as a MODEM setting option (e.g., via a new "AT" command).

As it stands (version `0.5.52c0a446`), additional incoming connections are successful, and data can be transmitted to the #FujiNet, but appear to be buffered somewhere along the way. Once the earlier active connection has ended (e.g., `exit`/`logout` on a shell you're TELNETing to, "`+++ATH`" to hang up), anything buffered by the subsequent incoming connection is immediately sent to the Atari by #FujiNet.

# Example
1. Boot into Ice-T on the _Atari_
2. Issue `ATPORT23` and `ATS0=1` to the #FujiNet to allow incoming connections to TELNET port (port 23).
3. Issue `ATNET1` (telnet mode) and `ATDTaaa.bbb.ccc.ddd` to the #FujiNet, to connect to a _Remote Host_ via TELNET.
4. Begin interacting with _Remote Host_; keep the connection alive.
5. On _Client_, TELNET to the #FujiNet (e.g., `telnet fujinet-ip-addr`)
6. The `telnet` client shows a successful connection, and things typed on _Client_ appear echoed back
7. Nothing typed in the `telnet` client on _Client_ appears on the _Atari_ (yet)
8. Type "`+++`" on the _Atari_ (or otherwise, e.g. issuing `exit` at a shell prompt), disconnecting the _Atari_'s connection to _Remote Host_ that was started in step 3
9. Everything typed on _Client_ in step 6 now appears on the _Atari_

# Motivation
There's no indication to the user connecting to the #FujiNet that their connection is not fully active at the Atari end. Back in the day, with a real modem and POTS phone line, the user would receive a busy signal and their modem will respond with "`BUSY`" (verbose) or "`7`" (Hayes modem numeric code).

Emulating this would seem wise, especially when #FujiNets start getting used as a means to connect real Atari BBSes up to the Internet.  (Note that my example above involves the Atari making an outgoing connection (Step 3), and an incoming connection (Step 5). However, the same happens -- and is relevant to a BBS situation -- if multiple incoming connections occur.)

billkendrick, 2020-11-16:
What I'd expect, trying to telnet to the #FujiNet while it was busy servicing another connection, would be e.g.:
```
$ telnet fuji-net-ip-addr
telnet: Unable to connect to remote host: Connection refused
```

tschak909, 2020-11-16:
Yup, basically this involves closing the listening socket once a connection
has been accepted.

-Thom

tschak909, 2020-11-27:
```
thomc@TMA-2:~$ telnet thom 6502
Trying 192.168.1.11...
Connected to thom.home.
Escape character is '^]'.
The MODEM is currently serving another caller. Please try again later.
^]
^CConnection closed by foreign host.
thomc@TMA-2:~$
```

commit 14b4b5a0 (2020-11-27): Fixes #397 - add busy message.
closed 2020-11-27 by tschak909