FujiNetWIFI/fujinet-firmware #399: If tnfsd returns invalid session ID, attempt to re-mount.

tschak909, 2020-11-18:
Since builds of tnfsd of 2020.1115 and later now have a 6 hour session time-out, computers using mounted TNFS resources and left alone for, say, a night, may find their sessions dead the next morning, forcing a reboot.

IMO, the correct approach here would be to check for an INVALID SESSION ID error, and re-mount the host if this happens, when attempting any TNFS operation.

-Thom

tschak909, 2021-02-11:
I'll take a whack at this. On it, now.

RichStephens, 2026-10-03:
This is done. When the server answers with an invalid session, `_tnfs_transaction()` in `lib/TNFSlib/tnfslib.cpp` now remounts (`_tnfs_session_recovery()`) and retries the command with the new session. On top of that, #727 added a keep-alive: while a TNFS host is mounted, FujiNet sends a small request every 60 seconds, so the idle timeout described here doesn't expire the session overnight anymore.

One limit: only commands that don't need an open handle (open, stat, opendir, mkdir and so on) can be retried after a remount. A read or write on a file that was already open gets `BAD_FILENUM`, because the handle went away with the old session. For mounted disk images it's the keep-alive that does the work.

closed 2026-10-03 by RichStephens