FujiNetWIFI/fujinet-firmware #406: Firmware: SD browse to directory issue

SpudsterZ, 2020-11-22:
**Machine**
`Atari 800 w/ Incognito set to basic Colleen mode.`

**Scenario**
```
Browsing to a particular directory:
1. White light solid, orange light solid
2. Pause
3. Both lights out
4. White light on
5. Longer pause
6. ![ERROR OPENING DIRECTORY](https://user-images.githubusercontent.com/74847414/99896101-dce70480-2c52-11eb-9e21-001226a849dc.png)

Must ESC back to main - all other options fail (N/F/DELETE/OPTION).

```

**Directory details**
```
338.1 MiB (354,516,013)
11,166 files, 0 sub-folders
```

**Feedback**
`Clearly an extreme case - minor issue. A more obvious error condition would be helpful. Gold star for forcing user to return back to main or recovering from error so other options function (i.e. up directory).`

RichStephens, 2026-10-03:
Still a problem as of current master, and it looks like a crash rather than just a failed open. `FileSystemSDFAT::dir_open()` reads the entire directory into `std::vector<fsdir_entry>`, about 272 bytes per entry because of the fixed 256-byte filename, sorts it, then copies it into `_dir_entries`. Nothing limits the count. 11,166 files is about 3 MB of entries, plus the vector's growth and the final copy, which is more than the ESP32's PSRAM. With exceptions disabled, the failed allocation aborts, which would explain the lights going out and FujiNet coming back. Fixing it means not caching every entry at full size, either by storing names compactly, capping the count, or streaming instead of sorting the whole directory.

RichStephens, 2026-10-16:
@tschak909 @FozzTexx Still real: `FileSystemSDFAT::dir_open()` caches every entry at 272 bytes, holds two copies while sorting, and has no limit, so a directory this size outgrows PSRAM and the failed allocation aborts (measured on FujiNet-PC with 11,166 files: peak +23 MB on the host-size entries). Proposed fix, three PRs: (1) compact storage inside the shared `DirCache` (one name arena plus 16-byte records, same `fsdir_entry*` API, a host test); (2) move SD onto `DirCache` instead of its own copy of the filter/sort code; (3) a budget check so an oversized directory returns an error instead of rebooting the device. `DirCache` is shared by every filesystem, so I'd like a yes from one of you before starting. If I don't hear back in a week I'll go ahead.

closed 2026-10-27 by tschak909