FujiNetWIFI/fujinet-firmware #412: The Fuji-kick-in bug is still present...

mr-atari, 2020-12-15:
The Fuji-kick-in bug is still present though....
So I boot from 1050, go to DOS.
Do "DIR" several times and after say 5-6 DIRs it kicks-in.
2 devices on D1, mayhem.
Can't access my 1050, again, very annoying.
When I switch off my 1050 and do DIR, I get the files present on the rom-config-disk.
Do I have a ticket number for this?
I would like to have this fixed, thanks.

tschak909, 2020-12-15:
even when you disable the disk from web admin?
-Thom

mr-atari, 2020-12-16:
Let me explain what I think the problem with the firmware is:
>Fuji counts STATUS to D1, right?
>When counter reaches xx ROM-disk kicks in, right?

You should cancel this routine (stop or reset), what ever you guys seem appropriate IMMEDIATE:
When: Another command then STATUS is send to D1

Doing IO or DIRs is sending STATUS to D1 to check density. Do this xx times and ROM-disk kicks in. Mayhem on SIO to D1.....

For your information, look at the 850, it counts STATUS to D1 for it's ROM-disk kick in too. But it cancels it when another command is received to D1. This stays OFF until the 850 is power-cycled.

mr-atari, 2020-12-16:
[2nd reply]
>even when you disable the disk from web admin?
No.
But that is not a workable situation too, since I am blocked out to have any kind of access to the ROM-disk....
Imagine having no web admin at the moment?

mr-atari, 2020-12-27:
Latest firmware (today)) bug still present.....
I simply can't understand why you guys are unable to reproduce.
Its very simple:
Do NOT boot from Fuji, do NOT mount anything, just boot over SIO from (any) diskdrive (real or emulated).
Just do a couple of status/dir to D1 and after 4-5 times, mayhem over SIO to D1:
The ROM-disk has kicked in.

https://user-images.githubusercontent.com/43779353/103168074-81e79500-4830-11eb-937e-9d9fe3b57e3c.mp4

tschak909, 2020-12-29:
Fixed.

closed 2020-12-29 by tschak909