FujiNetWIFI/fujinet-firmware #415: Cannot use micro SD card slot after flashing to 0.5.8edf468d 2020-12-19

8bitbill, 2020-12-20:
Hello,

I just flashed from the previous version to current (v0.5.8edf468d 2020-12-19).

FujiNet was completely reset back to defaults after this flash upgrade (I have not had that happen during previous flash upgrades).

But now since upgrading, if there is a card in the SD slot, FujiNet will not cold boot or reset (via it's reset button) successfully.
Debug output shows it is stuck in this error then rebooting loop:

[09:35:04]FujiNet 0.5.8edf468d 2020-12-19 04:49:28 Started @ 3
[09:35:04]Starting heap: 4053896
[09:35:04]PsramSize 3932108
[09:35:04]himem phys 4456448
[09:35:04]himem free 4456448
[09:35:04]himem reserved 262144
[09:35:04]FujiNet Hardware v1.0
[09:35:04]SPIFFS mounted.
[09:35:04]E (1116) sdmmc_sd: sdmmc_enable_hs_mode_and_check: select_card (1) returned 0x106
[09:35:04]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[09:35:04]Core 0 register dump:
[09:35:04]PC      : 0x4012e100  PS      : 0x00060f30  A0      : 0x8012e390  A1      : 0x3ffbb220  
[09:35:04]A2      : 0x3ffc75e4  A3      : 0x3ffbb2f9  A4      : 0x00000010  A5      : 0x3f800024  
[09:35:04]A6      : 0x00011ae8  A7      : 0x00000000  A8      : 0x8012f80d  A9      : 0x3ffbb210  
[09:35:04]A10     : 0x00000010  A11     : 0x00000000  A12     : 0x00011ae8  A13     : 0x3f811b0c  
[09:35:04]A14     : 0x40139cbc  A15     : 0x000011ae  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
[09:35:04]EXCVADDR: 0x8012e594  LBEG    : 0x4009151a  LEND    : 0x40091525  LCOUNT  : 0x00000000  
[09:35:04]
[09:35:04]ELF file SHA256: f95a223090c2d556
[09:35:04]
[09:35:04]Backtrace: 0x4012e0fd:0x3ffbb220 0x4012e38d:0x3ffbb2f0 0x400d34ed:0x3ffbb330 0x400d2b4a:0x3ffbb3c0 0x400d2c67:0x3ffbb3e0 0x4013e499:0x3ffbb410 0x40095bed:0x3ffbb430
[09:35:04]
[09:35:04]Rebooting...

If no card is inserted during either of those sequences, FujiNet will boot up and be usable, except for the SD card slot will not work and reports just Error on the Atari's screen.

Here is the debug output from the flash upgrade followed by the SD card slot error (because a SD card was in the slot)
I have tried with the two previously working SD cards I have and neither is working now with the SD slot.  They both still have their contents when I look at them via another machine.
I also attempted re-flashing with the same results.

Thank you for your time.
-Bill

Using '/dev/ttyUSB0' as serial port.
Getting latest firmware from fujinet.online..
FujiNet Version: 0.5.8edf468d
Version Date/Time: 2020-12-19 04:49:28
Build Date/Time: Sat Dec 19 19:09:59 UTC 2020
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ5 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: AC:67:B2:76:F3:B8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 16MB
 - Firware path: fujinet.online
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
[... 123 lines cut ...]
[09:34:55][09:34:55]
[09:34:55]--~--~--~--
[09:34:55]FujiNet 0.5.8edf468d 2020-12-19 04:49:28 Started @ 3
[09:34:55]Starting heap: 4053896
[09:34:55]PsramSize 3932108
[09:34:55]himem phys 4456448
[09:34:55]himem free 4456448
[09:34:55]himem reserved 262144
[09:34:55]FujiNet Hardware v1.0
[09:34:55]SPIFFS mounted.
[09:34:55]E (1116) sdmmc_sd: sdmmc_enable_hs_mode_and_check: select_card (1) returned 0x106
[09:34:55]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[09:34:55]Core 0 register dump:
[09:34:55]PC      : 0x4012e100  PS      : 0x00060f30  A0      : 0x8012e390  A1      : 0x3ffbb220  
[09:34:55]A2      : 0x3ffc75e4  A3      : 0x3ffbb2f9  A4      : 0x00000010  A5      : 0x3f800024  
[09:34:55]A6      : 0x00011ae8  A7      : 0x00000000  A8      : 0x8012f80d  A9      : 0x3ffbb210  
[09:34:55]A10     : 0x00000010  A11     : 0x00000000  A12     : 0x00011ae8  A13     : 0x3f811b0c  
[09:34:55]A14     : 0x40139cbc  A15     : 0x000011ae  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
[09:34:55]EXCVADDR: 0x8012e594  LBEG    : 0x4009151a  LEND    : 0x40091525  LCOUNT  : 0x00000000  
[09:34:55]
[09:34:55]ELF file SHA256: f95a223090c2d556
[09:34:55]
[09:34:55]Backtrace: 0x4012e0fd:0x3ffbb220 0x4012e38d:0x3ffbb2f0 0x400d34ed:0x3ffbb330 0x400d2b4a:0x3ffbb3c0 0x400d2c67:0x3ffbb3e0 0x4013e499:0x3ffbb410 0x40095bed:0x3ffbb430
[09:34:55]
[09:34:55]Rebooting...

tschak909, 2020-12-20:
The error returned from select_card is 0x106 which means feature not
supported. So the SDMMC controller isn't happy about something with your
card. Can you try another one?

I use a Sandisk 2GB card.

-Thom

8bitbill, 2020-12-20:
I have been using 2 cards exclusively with the FujiNet device -  a 2GB Sandisk and a 2GB Samsung with no issues  
And they were working yesterday with FujiNet under the previous firmware.
But they both fail now since upgrading to the current firmware. (12/19/20).  
I have a 32 GB Samsung card I was not using at all, I formatted that and it has the same issue with not working with the FujiNet.  
All of these cards do work on my Win and Mac systems.
Did I just get a bad flash of the device this go around?  Is there another way to flash a FujiNet that I should try?
Thank you.
-Bill

tschak909, 2020-12-20:
Try running the flasher program again? it will try to download the binary
again and re-flash the device.

The error you're getting should not be happening under any circumstance, as
we are not sending commands to the SD controller that aren't supported.
Also, nobody else has reported this issue.

-Thom

8bitbill, 2020-12-20:
FYI -
With the micro SD card slot empty (no card installed), I get this output when hitting the reset button on the FujiNet:
Note the SD error - is that the standard error response to receive when no card is installed?
Thanks,
-Bill
Using '/dev/ttyUSB0' as serial port.
Showing logs:
[13:22:32][13:22:32]
[13:22:32]--~--~--~--
[13:22:32]FujiNet 0.5.8edf468d 2020-12-19 04:49:28 Started @ 3
[13:22:32]Starting heap: 4053896
[13:22:32]PsramSize 3932108
[13:22:32]himem phys 4456448
[13:22:32]himem free 4456448
[13:22:32]himem reserved 262144
[13:22:32]FujiNet Hardware v1.0
[13:22:32]SPIFFS mounted.
[13:22:32]E (1128) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
[13:22:32]SD mount failed with code #264, "ESP_ERR_INVALID_RESPONSE"
[13:22:32]fnConfig::load
[13:22:32]fnConfig::load read 451 bytes from config file
[13:22:32]WIFI_EVENT_STA_START
[13:22:32]WiFi connect attempt to SSID "private"
[13:22:32]esp_wifi_connect returned 0
[13:22:32]disk MOUNT
[13:22:32]ATR MOUNT
[13:22:32]mounted ATR: paragraphs=5760, sect_size=128, sect_count=720, disk_size=0
[13:22:32]Creating a default printer using FS_SPIFFS storage and type 1
[13:22:32]SSH client initialized.
[13:22:32]ModemSniffer::ModemSniffer(FS_SPIFFS)
[13:22:32]SIO SETUP
[13:22:32]Set HSIO baud from 0 to 67431 (index 6), alt=68209
[13:22:32]Available heap: 3985324
[13:22:32]Setup complete @ 410 (407ms)
[13:22:32]WIFI_EVENT_STA_CONNECTED
[13:22:33]IP_EVENT_STA_GOT_IP
[13:22:33]Obtained IP address: 172.16.0.148
[13:22:33]SNTP client start
[13:22:33]Starting web server on port 80
[13:22:36]SNTP time sync event: Sun Dec 20, 13:22:36 2020 -0600

tschak909, 2020-12-20:
Yes, that's fine. I'm definitely confused. Is the SD card slot perhaps
intermittently connected?

-Thom

tschak909, 2020-12-20:
I have reached out to other users to see if they've seen the same problem.

In the mean time, perhaps contact vendor to see if you can swap them out?

-Thom

8bitbill, 2020-12-20:
ok - thanks.

mozzwald, 2020-12-27:
@8bitbill I pushed another update yesterday. Can you try flashing your FujiNet again with the latest?

8bitbill, 2020-12-27:
Sure.  I just flashed it with that build and now my micro-SD cards are working again.
Thank you!
Can I ask - was there something you explicitly changed to affect this?

mozzwald, 2020-12-28:
> was there something you explicitly changed to affect this?

Nope :) I'm at a loss as to what caused it. 

_“There was a fish in the percolator.”_

closed 2020-12-28 by mozzwald