FujiNetWIFI/fujinet-firmware #417: Airplane Mode

TheRobotFactory, 2020-12-21:
Is it possible to allow the Fujinet to start up and load the config program even if there is no network available or if the user doesn't want to connect to a network for some reason. Presently if there is no network the device gets stuck at the connection screen.

I'd suggest holding down a button or a combination of buttons at startup (or selecting the option from the web interface/config app page) to persistently toggle wifi as disabled/enabled so that the config app knows to ignore or wait for the connection step and just display the SD host device or the full list of hosts.

mozzwald, 2020-12-27:
This is an issue with CONFIG rather than the firmware so I have opened an issue there. CONFIG should have a way to skip WiFi connection. The firmware doesn't really care if you connect or not and will run fine without a connection. Firmware should just throw errors if you try to access/mount a network disk, etc.

https://github.com/FujiNetWIFI/fujinet-config/issues/24

tschak909, 2020-12-27:
This issue has been fixed in the above ticket.
[image]

closed 2020-12-27 by tschak909