FujiNetWIFI/fujinet-firmware #421: CAS files in slot 8 show error in CONFIG during mount

mozzwald, 2021-01-02:
CAS files in disk slot 8 throw an error when pressing OPTION to mount & boot. Maybe the firmware should pass a SUCCESS and skip actually trying to mount if the file extension is CAS.

```
[14:25:40]CF: 70 f8 07 01 71
[14:25:40]sioFuji::sio_process() called
[14:25:40]ACK!
[14:25:40]Fuji cmd: MOUNT IMAGE
[14:25:40]Selecting '/O'Riley's Mine.cas' from host #2 as r on D8:
[14:25:40]fujiHost #2 opening file path "/O'Riley's Mine.cas"
[14:25:40]ERROR!
```

tschak909, 2021-01-15:
@mozzwald is this still happening for you?

are you doing it on an SD slot? I'll test.

16:20:16.327 > CF: 70 f9 03 00 6d
16:20:16.327 > sioFuji::sio_process() called
16:20:16.327 > ACK!
16:20:16.327 > Fuji cmd: MOUNT HOST
16:20:16.327 > ::mount {3} "atari-apps.irata.online"
16:20:16.327 > ::mount_local Attempting mount of "atari-apps.irata.online"
16:20:16.338 > ::mount_tnfs {3:2} "atari-apps.irata.online"
16:20:16.338 > ::mount_tnfs Currently connected to host "atari-apps.irata.online"
16:20:16.338 > COMPLETE!
16:20:16.350 >
16:20:16.355 > CF: 70 f8 07 00 70
16:20:16.355 > sioFuji::sio_process() called
16:20:16.355 > ACK!
16:20:16.355 > Fuji cmd: MOUNT IMAGE
16:20:16.355 > Selecting '/Games/a8p/Baseball (1982)(Aim Software)(US)[CLOAD+RUN][BASIC].cas' from host #3 as r on D8:
16:20:16.355 > fujiHost #3 opening file path "/Games/a8p/Baseball (1982)(Aim Software)(US)[CLOAD+RUN][BASIC].cas"
16:20:16.392 > TNFS open file: "/Games/a8p/Baseball (1982)(Aim Software)(US)[CLOAD+RUN][BASIC].cas" (0x0001, 0x01b6)
16:20:16.425 > File opened, handle ID: 0, size: 15276, pos: 0
16:20:16.425 > ::get_filesize
16:20:16.425 > tnfs_lseek currpos=0, pos=0, typ=1
16:20:16.460 > tnfs_lseek success, new pos=0, response pos=0
16:20:16.494 > tnfs_lseek currpos=0, pos=0, typ=2
16:20:16.537 > tnfs_lseek success, new pos=15276, response pos=15276
16:20:16.572 > tnfs_lseek currpos=15276, pos=0, typ=0
16:20:16.610 > tnfs_lseek success, new pos=0, response pos=0
16:20:16.610 > disk MOUNT
16:20:16.610 > Cassette image filesize = 15276
16:20:16.612 > tnfs_lseek currpos=0, pos=0, typ=0
16:20:16.651 > tnfs_lseek success, new pos=0, response pos=0
16:20:16.709 > FUJI File Found
16:20:16.709 > COMPLETE!

mozzwald, 2021-01-15:
> @mozzwald is this still happening for you?
> 
> are you doing it on an SD slot? I'll test.

I just tested with CAS file from tnfs and SD. Problem is not there anymore.

closed 2021-01-15 by mozzwald

tschak909, 2021-01-15:
. o O (am wondering if I fixed it at some point?) ;)