FujiNetWIFI/fujinet-firmware #424: Crashing after update to Jan 16th firmware

8bitbill, 2021-01-17:
After upgrading to the recent Jan 16th firmware release (no errors during flashing), when the FN rebooted, it resulted in this endless loop when attempting to store the SD's fnconfig.ini it found to the SPIFFS:

_[11:11:53]Rebooting...
[11:11:53][11:11:53]
[11:11:53]--~--~--~--
[11:11:53]FujiNet 0.5.e09997c5 2021-01-16 02:23:59 Started @ 3
[11:11:53]Starting heap: 4053880
[11:11:53]PsramSize 3932108
[11:11:53]himem phys 4456448
[11:11:53]himem free 4456448
[11:11:53]himem reserved 262144
[11:11:53]FujiNet Hardware v1.0
[11:11:53]SPIFFS mounted.
[11:11:53]SD mounted.
[11:11:53]fnConfig::load
[11:11:53]Found copy of config file on SD - copying that to SPIFFS
[11:11:53]copy_file "/fnconfig.ini" -> "/fnconfig.ini"
[11:11:53]Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
[11:11:53]Core 0 register dump:
[11:11:53]PC      : 0x4012f034  PS      : 0x00060730  A0      : 0x80166d9a  A1      : 0x3ffbb1d0  
[11:11:53]A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
[11:11:53]A6      : 0x3f800040  A7      : 0x00000001  A8      : 0x8012eff4  A9      : 0x3ffbb160  
[11:11:53]A10     : 0x00000000  A11     : 0xc1a67ebb  A12     : 0xc1a67ebb  A13     : 0x3f80188c  
[11:11:53]A14     : 0x3f8018b4  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
[11:11:53]EXCVADDR: 0x00000000  LBEG    : 0x40092e7d  LEND    : 0x40092e8d  LCOUNT  : 0xfffffffc  
[11:11:53]
[11:11:53]ELF file SHA256: 05cb431088309a89
[11:11:53]
[11:11:53]Backtrace: 0x4012f031:0x3ffbb1d0 0x40166d97:0x3ffbb200 0x4021d005:0x3ffbb220 0x4021d0c1:0x3ffbb250 0x400d32f0:0x3ffbb270 0x400e9b7e:0x3ffbb290 0x400d6806:0x3ffbb2b0 0x400d2b53:0x3ffbb3c0 0x400d2c87:0x3ffbb3e0 0x4013e751:0x3ffbb410 0x40095bed:0x3ffbb430
[11:11:53]
[11:11:53]Rebooting...
[11:11:54][11:11:54]_

The contents of the fnconfig.ini on the SD card that it was attempting to copy to SPIFFS:

_[General]
devicename=FujiNet
hsioindex=0
rotationsounds=1
configenabled=1
timezone=CST+6CDT

[WiFi]
SSID=private
passphrase=private

[Network]
sntpserver=pool.ntp.org

[Host1]
type=SD
name=SD

[Host2]
type=SD
name=FujiNet.online

[Host3]
type=SD
name=atari-apps.irata.online

[Host4]
type=TNFS
name=FujiNet.pl

[Mount1]
hostslot=4
path=/uploads/Flickerterm.atr
mode=r

[Printer1]
type=17
port=1

[Modem]
sniffer_enabled=0

[Cassette]
play_record=0 Play
pulldown=0 B Button Press_

tschak909, 2021-01-17:
stray underscore character in fnConfig.ini. Remove the _ before [General]
-Thom

8bitbill, 2021-01-17:
Sorry, typo when pasted into github - there was no '_'  in that original file.

8bitbill, 2021-01-18:
I downgraded my FN firmware back to the previous version, v0.5.89f318f9, and regained functionality to my FN, no more StoreProhibited errors with the previous firmware.

tschak909, 2021-01-19:
Can you try upgrading again, but deleting the fnConfig.ini ?

We are rapidly upgrading the device, and keeping it at a specific version is a very bad idea.
-Thom

tschak909, 2021-01-25:
Any update on this? We've done two pushes of firmware since.
-Thom

closed 2021-03-19 by tschak909