FujiNetWIFI/fujinet-firmware #433: Refactor MODEM code

tschak909, 2021-02-09:
The current WiFi MODEM code lives monolithically inside sio/modem.cpp.

This should be re-factored outward into its own library under lib/ so it can be used by other peripheral buses (and by CP/M)

The idea here is to:

* Seperate out the core modem code
* Call an initialization function with a pointer to the desired UART.
* Replace explicit calls to fnUARTSIO to use this pointer.
* Make lib/sio/modem.cpp use the now factored library.

brad-colbert, 2021-04-15:
It this simply a "move and create an interface" job?

tschak909, 2021-04-16:
yeah.

dillera, 2026-10-03:
This looks to have been addressed by:

- #528 move modem into lib/ and refactor to use it with new uart member (merged 2022-11-20)

Closing as completed. If you can still reproduce this on a current firmware build, please reopen it (or open a new issue) with the firmware version, the platform, and the steps.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera