FujiNetWIFI/fujinet-firmware #438: [N:] HTTP Adapter needs Chunked support

tschak909, 2021-03-19:
The N: HTTP adapter needs support added for HTTP Chunked traffic:

The relevant code:
Protocol adapter:
https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/HTTP.cpp

HTTP class:
https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/http/fnHttpClient.cpp

Currently, the existing code assumes that the body size is known. When the data is chunked, the content size is -1 bytes, so when the data is chunked, we need to get the size of the current chunk, and return it into the RX buffer.

http://www.twitter.com/ and http://www.google.com/ can be used as test URLs to make sure that this works properly.

brad-colbert, 2021-04-15:
Are all the instances where  `actual_len = client->read((uint8_t *)buf, len);`  is called, is this where this needs to be handled?

tschak909, 2021-04-16:
at least.
-Thom

brad-colbert, 2021-04-16:
Anything more than filling buff until actual_len == len?

tschak909, 2021-04-16:
0Not sure, investigation needed.

Basically, the relevant API is here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_client.html

and the esp_http_client_get_chunk_length call will ultimately help here.

tschak909, 2021-07-25:
Since @mozzwald implemented this, closing it.

closed 2021-07-25 by tschak909