FujiNetWIFI/fujinet-firmware #439: Add SSH protocol adapter

tschak909, 2021-03-26:
Now that libssh2 is embedded, a protocol adapter can be made to be able to do interactive shell sessions:

To implement:

[ ] copy https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/Test.cpp and https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/Test.h to SSH.cpp and SSH.h in lib/network-protocol
[ ] implement functions (open, close, read, write, status)
[ ] use netcat (e.g. on atari-apps/Networking TNFS) to test connection with a host

Existing protocol code can be seen from the above two links, as well as https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/TCP.cpp and https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/TCP.h

In addition, there is plenty of example code with libssh2, here is a complete ssh2 client that opens an interactive terminal:
https://www.libssh2.org/examples/ssh2.html

markjfisher, 2023-06-27:
Is this complete?

tschak909, 2023-06-27:
It works for me, it definitely needs more error checking, but, to use, you send both a LOGIN and PASSWORD command with the appropriate credentials.

https://github.com/FujiNetWIFI/fujinet-apps/blob/master/netcat/atari/src/main.c#L63

closed 2023-07-25 by tschak909