FujiNetWIFI/fujinet-firmware #443: [insert-libssh branch] libssh isn't compiling correctly.

tschak909, 2021-04-17:
In the insert-libssh branch, I have replaced libssh2, with a libssh component, now living in components/libssh.

Unfortunately, it doesn't seem to be building with the current esp-idf that's provided with platform.io, giving the following linking error:

```
Linking .pio\build\fujinet-v1\firmware.elf
c:/users/thomc/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\esp-idf\libssh\liblibssh.a(chachapoly.o):(.literal.chacha20_poly1305_aead_decrypt+0x8): undefined reference to `secure_memcmp'
c:/users/thomc/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\esp-idf\libssh\liblibssh.a(chachapoly.o): in function `chacha20_poly1305_aead_decrypt':
C:\Users\thomc\Workspace\fujinet-platformio/components\libssh\libssh\src/chachapoly.c:157: undefined reference to `secure_memcmp'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\fujinet-v1\firmware.elf] Error 1
```

The relevant test code is already in lib/network-protocol/SSH.cpp to initialize the session and free it, to make sure the library does compile in.

markjfisher, 2022-11-20:
@tschak909 can this be closed?

tschak909, 2022-11-20:
yup.

closed 2022-11-20 by tschak909