FujiNetWIFI/fujinet-firmware #444: We need an XPath parser for XML.

tschak909, 2021-04-17:
We need an XPath parser of some sort that can be attached to any protocol (e.g. HTTP).

I did a video demonstration showing a possible interface for the XML parser, here:
https://www.youtube.com/watch?v=0C23ffVWvkM This was done with older code, and was just a proof of concept

But maybe this is a jumping off point for a better idea?

-Thom

idolpx, 2021-04-18:
TinyXML  or maybe RapidXML would do the trick if they will compile for ESP32.

https://github.com/leethomason/tinyxml2
http://rapidxml.sourceforge.net/index.htm

idolpx, 2021-04-18:
I don't think XPath is implemented in those two though. :(

idolpx, 2021-04-18:
This might do the trick with TinyXML2 though. https://github.com/stanthomas/tinyxml2-ex

RichStephens, 2026-10-03:
This is done. #1557 added an XML channel mode (`lib/fnxml/`), using tinyxml2 as suggested above, with a query engine for a subset of XPath: `/` steps, `//` descendants, `*`, `[n]`, `[@attr]`, `[@attr="v"]`, `[text()="v"]` and a trailing `@attr`. It works over any protocol the channel opens, HTTP included, and has unit tests. Full XPath (other axes and functions) isn't there; if something specific is missing, please open a new issue for it.

closed 2026-10-03 by RichStephens