FujiNetWIFI/fujinet-firmware #448: Problem accessing APETIME when booting a program over SIO2SD with high speed IO

billkendrick, 2021-05-02:
When booting the APOD viewer client (2021-05-01 version), either on an ATR disk image (containing MyDOS and the program) or as an XEX directly, using an SIO2SD with high speed I/O (the config on mine shows `sio high speed: 59659.08 ($08)`), the APOD client cannot _ever_ properly fetch the current time & date via FujiNet's APETIME protocol.

Unlike what happens when either (a) FujiNet is detached or turned off, or (b) the APOD client is booted so quickly that FujiNet is not yet up and running (I generally use my The Ultimate Cart SD cartridge, as it's the quickest way to fire up the XEX), which seems to be APETIME protocol coming back with an empty string, it instead seems to return some kind of gibberish. (What I've seen the APOD viewer client parsing it as were either "2000-00-15", or "2000-120-15".)

Based on this, I added extra sanity checks in APOD client (https://github.com/FujiNetWIFI/fujinet-apps/commit/f0d7212d32d2e5dda4466dd266cde77abb967f48).

RichStephens, 2026-10-03:
Closing this one for now. The report is from May 2021, and much of the clock code has been reworked since then: the time commands, APETIME included, now live in a shared clock class (`lib/device/fujiClock/`) instead of per-bus code, so I can't tell whether this still happens.

@billkendrick, could you try again with the current firmware? If APETIME still returns a bad date or time when booting over a high-speed SIO2SD, please open a new issue with what you see, and the FujiNet log if you can capture it.

We don't have a high-speed SIO2SD setup to test this ourselves, so if it's still a problem, fixing it would need your help trying builds and reporting back.

closed 2026-10-03 by RichStephens

billkendrick, 2026-10-03:
I'm not in a position to test. My 8-bit has no display, so I'm unable to use it right now. Prior to that, I got out of the habit of using my SIO2SD when doing any FujiNet activity. If I happen to get in this situation again, I'll try to remember this ticket already exists, and reopen it. But for now, since no one else seemed to chime in, it's obviously hasn't been a big issue. :)