FujiNetWIFI/fujinet-firmware #449: APETIME enhancement: fetch the FujiNet's clock's timezone, and/or fetch the time in a given timezone

billkendrick, 2021-05-03:
It'd be handy to have FujiNet tell me what timezone it's in, or the ability to have it convert time to a given timezone.

Motivation: The APOD viewer client allows you to pick a date, up to "the current day".  So right now, since it is 11pm on May 2nd, 2021 where I'm at (Pacific time zone), the maximum date the app lets me pick is 2021-05-02.

However, the APOD website is on an Eastern timezone, and therefore where they are, it's already May 3rd, and therefore a new image is available.  Now, while I cannot pick "2021-05-03" in the APOD viewer client on my Atari, I can at least tell the client "just get the latest" (no date specified -- so it fetches from the APOD site's main page, rather than a date-stamped archive page).  Users east of the Eastern time zone (e.g., in Europe, etc.) might end up picking a date "in the future", which would cause the server to fail, and no image to appear.

Therefore, it'd be very handy if I could, in the APOD viewer app, know what day it is _in the Eastern time zone_, so as to avoid confusing users in the other 23 (more, actually) time zones around the world. :wink: 

Perhaps a separate request type could be invented that provides something more than YYMMDD HHMMSS results that `SIO_APETIMECMD_GETTIME` provides.  I'm unclear what, if anything, other systems might provide ([APE](https://www.atarimax.com/ape/ape.htm) itself, [RespeQt](https://github.com/RespeQt/RespeQt), [AtariSIO](https://github.com/HiassofT/AtariSIO), etc.)

Admittedly, this is a pretty specialized request, and there's a workaround in the APOD viewer app itself... so consider this a low-priority (or feel free to close as 'wontfix'... though I do kind of wonder what other network-based apps might benefit from knowing the true time, say in UTC.  Say, a game where high scores are only accepted up until a particular point in time.)

mozzwald, 2021-06-19:
This should be easy enough to do in the firmware by adding a new command. You can duplicate the existing function, then change the command code and [one line](https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/sio/apetime.cpp#L12) to [get UTC instead of local time](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html#get-current-time). Would you like to take a stab at it?

billkendrick, 2021-06-20:
Thanks for the details, @mozzwald. Sure, I can take a stab at it at some point soon.

mozzwald, 2021-07-17:
Excellent work, thanks!

fixed with bbbb9f966313c5948151c14512eb24d08fe36288

closed 2021-07-17 by mozzwald