FujiNetWIFI/fujinet-firmware #452: IRAM0 segment data does not fit

monsoft, 2021-06-07:
Hi,

I tried to compile code using VSC and PlatformIO extension by following information on page https://github.com/FujiNetWIFI/fujinet-platformio/wiki/Board-bring-up-for-FujiNet-Platform.IO-code. All went well until linking started and I got error:
```
Linking .pio/build/fujinet-v1-8mb/firmware.elf
/home/irekp/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/fujinet-v1-8mb/firmware.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/irekp/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/irekp/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 196 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/fujinet-v1-8mb/firmware.elf] Error 1
```
I was attempting to build firmware for "fujinet-v1-8mb" by setting variable "default_envs = fujinet-v1-8mb"

monsoft, 2021-06-14:
Looks like issue is related to new version 3.2.1 of Espressif32 platform. When downgraded to version 3.2.0, all compilation was finished correctly.

mozzwald, 2021-06-19:
The new esp32 platform has increased memory usage. You can disable bluetooth in your platformio.ini file to get it to build or as you figured out, downgrade the platform.

closed 2021-10-03 by monsoft