FujiNetWIFI/fujinet-firmware #453: enable/disable SIO2BT via web config

marcowy, 2021-06-11:
I have custom PCB without buttons. It would be helpful to have a such option.

mozzwald, 2021-06-11:
This should be possible, will look into it.

mozzwald, 2021-06-11:
Now that I think about it, this may be difficult in practice. You would be able to turn on SIO2BT from the web interface, but when you do Wifi is disabled and Bluetooth enabled. I'm not sure if both can be enabled at the same time.

mozzwald, 2021-06-19:
> I have custom PCB without buttons. It would be helpful to have such an option.

I can add the ability to enable SIO2BT mode from the web admin but you will not be able to turn it off without removing the sd card and editing fnconfig.ini to disable SIO2BT mode. Will this work for you? Maybe you should just add buttons to your hardware.

marcowy, 2021-06-19:
I've added the button. Thanks!

closed 2021-06-25 by mozzwald