FujiNetWIFI/fujinet-firmware #454: Updates/new content for Quickstart Guide

spicyjack, 2021-06-18:
Hi,

I'd like to submit two patches for the Quickstart page on the wiki;

1. Add a link to the CONFIG page instructions in the "CONFIG" section on the Quickstart page
2. Add a section to the Quickstart page about using micro-SD cards with FujiNet, including not using _ExFAT_ with SD cards because it's not supported right now with FujiNet firmware, and size limits on SD Cards that you can use with FujiNet

[fujinet-quickstart-add_second_CONFIG_doc_link.patch.txt](https://github.com/FujiNetWIFI/fujinet-platformio/files/6674225/fujinet-quickstart-add_second_CONFIG_doc_link.patch.txt)
[fujinet-quickstart.sd_card_block.patch.txt](https://github.com/FujiNetWIFI/fujinet-platformio/files/6674226/fujinet-quickstart.sd_card_block.patch.txt)

Both patches should apply cleanly in any order, let me know if you run into any problems.  They are formatted patch files, despite the `*.txt` filename extensions.

tschak909, 2021-06-18:
Awesome! Just looked it over and looks okay. I'll apply in the next day or
two! :)

-Thom

spicyjack, 2021-11-10:
These changes were merged, closing this issue.

closed 2021-11-10 by spicyjack