FujiNetWIFI/fujinet-firmware #455: Feature Request: Cache Read Only ATR images from TNFS server to SD Card

mozzwald, 2021-06-29:
This could be done just like ATX images are and would/should only apply to read only disks on tnfs hosts and if SD Card is present. Some games and demos may constantly (re)read from the disk. This would save some bandwidth.

tschak909, 2021-06-29:
sounds good to me, will investigate.
-Thom

RichStephens, 2026-10-03:
Still wanted, as far as I can tell: nothing caches ATR images from TNFS.

There is something to start from now. `lib/FileSystem/fnFileCache.*` keeps a file in memory and moves it to an SD cache file past a size threshold. The HTTP and FTP filesystems use it, but TNFS and the Atari ATR media code don't. Open questions for a TNFS version: how to know the image is read-only, how to notice that the server's copy changed, and how to limit the space used on the SD card.