FujiNetWIFI/fujinet-firmware #461: Web interface section for reviewing and managing HTTP cookies

billkendrick, 2021-08-06:
On Discord, discussion of cookie handling on #FujiNet came up, and atariorbit suggested that we might want to expand the web-based management UI to allow management of cookies on the device.

This would assume some standardized method for storing cookies, which would likely be handled on a per-app basis, and stored in appdb storage (see e.g. https://github.com/FujiNetWIFI/fujinet-platformio/wiki/SIO-Command-$DC-Open-App-Key), so this may be difficult or impossible, or possibly moot.

dillera, 2026-10-03:
This issue has had no activity for about 5 years, and there have been 4357 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

billkendrick, 2026-10-03:
I admit, I have not thought about this since creating the ticket. 🤷  Sorry!

closed 2026-10-03 by billkendrick