FujiNetWIFI/fujinet-firmware #462: Firmware Upgrade Error: ESP Chip Auto-Detection failed

Sasatari, 2021-09-17:
Out of 20 FUjiNets V1.5 from Gavin, I get this when trying to update the firmware:

....._____....._____....._____....._____....._____Using '/dev/cu.usbserial-1120' as serial port.
Starting firmware upgrade...
Getting latest firmware from fujinet.online..
..FujiNet Version: 0.5.ba2f6751
Version Date/Time: 2021-07-27 12:03:04
Build Date/Time: Mon Aug 16 01:45:57 UTC 2021
...Connecting..._._._._._......______._._._._.
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Timed out waiting for packet header

It also renders the FujiNet unusable. When plugged into an Atari, it simply does nothing and the Atari boots to BASIC or SELF-TEST.

tschak909, 2021-09-17:
Try holding down A button when powering on. This will explicitly put the
ESP32 into flashing mode, then try the flash again.
-Thom

Sasatari, 2021-09-17:
This is what we do. But get this error.

Sasatari, 2021-09-17:
The best result we get is this:

Compressed 2097152 bytes to 648681...
Writing at 0x00cac000... (100 %)Wrote 2097152 bytes (648681 compressed) at 0x00c10000 in 15.6 seconds (effective 1078.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:

---

But it doesn´t show logs (like it does on the 17 working devices) and does not work.

tschak909, 2021-09-17:
Ok, looking at the error even further, this is even more odd, it's stating that the image that is downloaded is somehow corrupt?

hmm.. @mozzwald @jeffpiep ??

Sasatari, 2021-09-17:
Meanwhile I grabbed another board (still have 30 left), where it worked as expected.

Sasatari, 2021-09-17:
Complete Update log, after "Showing logs:" nothing more is displayed, nothing more happens, device doesn´t work on Atari. It´s just like it is not connected:

Using '/dev/cu.usbserial-1120' as serial port.
Starting firmware upgrade...
Getting latest firmware from fujinet.online..
FujiNet Version: 0.5.ba2f6751
Version Date/Time: 2021-07-27 12:03:04
Build Date/Time: Mon Aug 16 01:45:57 UTC 2021
Connecting.......Serial port closed!

Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ5 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: C4:5B:BE:38:48:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 16MB
 - Firware path: fujinet.online
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.7s
Compressed 27264 bytes to 16573...
Writing at 0x00005000... (100 %)Wrote 27264 bytes (16573 compressed) at 0x00001000 in 0.4 seconds (effective 529.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1344.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5056.3 kbit/s)...
Hash of data verified.
Compressed 2243936 bytes to 1299965...
Writing at 0x0014c000... (100 %)Wrote 2243936 bytes (1299965 compressed) at 0x00010000 in 30.6 seconds (effective 586.2 kbit/s)...
Hash of data verified.
Compressed 2097152 bytes to 648681...
Writing at 0x00cac000... (100 %)Wrote 2097152 bytes (648681 compressed) at 0x00c10000 in 15.6 seconds (effective 1072.5 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:

Sasatari, 2021-09-17:
Sidenote: There´s an "m" missing: Firware path: fujinet.online

Sasatari, 2021-09-17:
After switching the power-button, the logs showed up, but device still doesn´t work. As the serial log shows me all button presses, but no activity when turning the Atari on, I´ve asked Sleeπ to resolder all pins, as I suspect that the SIO connection might be interrupted.

iCityWork, 2021-09-17:
strange they all flashed fine before they were shipped. Everyone was flashed, SD card tested and WWW interface verified.

mozzwald, 2021-09-17:
@Sasatari what system/OS are you using to flash? have you tried another system and usb cable?

Sasatari, 2021-09-17:
I used OS X Big Sur on 13" MBP M1, Stefan Höhl used Xubuntu 20.04 LTS , Marc Brings used Win 7, Uli van Aspern Win 10.
All different cables, all with the same result.

mozzwald, 2021-09-17:
@iCityWork can you confirm that R5 and R6 are 10k?

iCityWork, 2021-09-17:
> @iCityWork can you confirm that R5 and R6 are 10k?

Absolutely yes.

Sasatari, 2021-09-17:
Confirming: R5 and R6 are 10K

mozzwald, 2021-09-18:
@iCityWork Which WROVER modules did you use? B, E, IE, I

iCityWork, 2021-09-18:
@mozzwald I believe that set was B.  As with the last batch of 75 that I just made. Again, all working as expected.

mozzwald, 2021-09-18:
> After switching the power-button, the logs showed up, but device still doesn´t work. As the serial log shows me all button presses, but no activity when turning the Atari on, I´ve asked Sleeπ to resolder all pins, as I suspect that the SIO connection might be interrupted.

You must have the power switch ON (UP) while flashing. Flipping the switch off and on is effectively a hard reset. 

How many boards are you having trouble with?

iCityWork, 2021-09-18:
Yes it must be turned on.  Again  all of them flashed before I sent them.  There were zero problems.  So not sure what’s going on but all the USB connectors worked as did all the power switches. I’m at a loss as to what the problem could be other than cables, drivers or who knows what.

mozzwald, 2021-09-18:
@Sasatari have you tried building from source and flashing with PlatformIO? I think it uses a different (newer) version of the esp32 flash tool.

Sasatari, 2021-09-18:
To be clear:
Gavins boards work perfectly well, as they are!
But as soon as one tries to update the firmware, _some_ of them render unusable. I've written in the ABBUC forums, that they can use it without hassle, but shall not attempt to update the firmware, until we tell them that it is safe.
Traveling to Vienna now (without Atari), "Bitbastel" took two bad boards home and will keep examining.

mozzwald, 2021-09-18:
> Gavins boards work perfectly well, as they are!
But as soon as one tries to update the firmware, _some_ of them render unusable.

Something is not right, what is wrong I do not know. I'm trying to narrow things down with the questions. If you want to send me a defective board or two I can do some testing here.

mozzwald, 2021-12-28:
Closing due to inactivity