FujiNetWIFI/fujinet-firmware #465: Feature Request: Filter directories

aptonline, 2021-09-30:
I've noticed a lot of tnfs hosts structure their content in directories for example:

`Server/Games/Game_Name/Game_File.atr
`
As the filter option does not filter directories this becomes slow and laborious to locate specific files in large directory structures.

As a FujiNet user it would be useful to have the ability to filter directories as well as files.

BONUS FEATURE: Is it possible to search recursively through all directories from the root of a host to find a file?

billkendrick, 2022-08-16:
It appears this would have to be supported by the TNFS server on the other end, although internally inside PlatformIO, it'd need to be replicated to support the same functionality when browsing a local SD card.

trekawek, 2025-02-23:
Filtering directories is supported, but the filter needs to end with a slash, e.g.:

```
Game_Name*/
```

> BONUS FEATURE: Is it possible to search recursively through all directories from the root of a host to find a file?

This has been contributed yesterday :) Please check: https://github.com/FujiNetWIFI/fujinet-firmware/discussions/845

aptonline, 2025-02-24:
> Filtering directories is supported, but the filter needs to end with a slash, e.g.:
>
> ```
> Game_Name*/
> ```
>
>
> This has been contributed yesterday :) Please check: [...]

Great news, thanks for letting me know :)