FujiNetWIFI/fujinet-firmware #471: Backup tapes through Fujinet

marsieman, 2021-11-13:
When attach a 1010 drive to the fujinet a CLOAD ends up with an "error 143 at line 0"
I guess fujinet interferes while using the tapedrive.
Meaning of all is saving old basic programm to the SD in the Fujinet
Maybe an option in the config to disable tape activity in the fujinet?

mozzwald, 2021-12-30:
I'm not sure if this will work, but try setting the "Activate by Pulldown Resistor or Button B?" option to Button B in the web interface and test if it disables the fujinet cassette interface.

[image]

tschak909, 2022-03-17:
@marsieman any update on this?

marsieman, 2022-03-17:
No still no solution for me.
the  "Activate by Pulldown Resistor or Button B?"  does not work for me.
Hot swapping works sometimes but not a good solution.
Thank you for looking into it  👍

mozzwald, 2022-03-25:
@marsieman I've pushed an update to the firmware that allows disabling cassette emulation from the webui. Can you please test and confirm this works? thanks

https://atariage.com/forums/topic/313303-fujinet-firmware-update-information-please-do-not-reply/?do=findComment&comment=5027820

dillera, 2026-10-03:
More information was asked for here about 5 years ago and there has been no reply, so closing this for now. If you can still reproduce this on a current firmware build, please reopen it (or open a new issue) with the firmware version, the platform, and the steps.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera