FujiNetWIFI/fujinet-firmware #479: B Button Double Tap Crash

iCityWork, 2021-12-24:
In the latest firmware, I am getting a crash when I double-tap the B button.  Used to get some diagnosis when double tapping B.

[11:21:08]BUTTON_B: DOUBLE-TAP
[11:21:08]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[11:21:08]
[11:21:08]Core  0 register dump:
[11:21:08]PC      : 0x400ef46a  PS      : 0x00060730  A0      : 0x800f0b4b  A1      : 0x3ffd0f40  
[11:21:08]A2      : 0x00000001  A3      : 0x0000000d  A4      : 0x3ffe2d00  A5      : 0x00000590  
[11:21:08]A6      : 0x00000003  A7      : 0x00060023  A8      : 0x3ffe2d00  A9      : 0x8295a964  
[11:21:09]A10     : 0x43552094  A11     : 0x0000fb01  A12     : 0x3ffcb5a8  A13     : 0xb33fffff  
[11:21:09]A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
[11:21:09]EXCVADDR: 0x8295a964  LBEG    : 0x4008e21d  LEND    : 0x4008e22d  LCOUNT  : 0xffffffff  
[11:21:09]
[11:21:09]Backtrace:0x400ef467:0x3ffd0f40 0x400f0b48:0x3ffd0f70 0x40095159:0x3ffd0fa0
[11:21:09]
[11:21:09]
[11:21:09]ELF file SHA256: 4ca0e402ce6de7e3
[11:21:09]
[11:21:09]Rebooting...
[11:21:09]
[11:21:09]
[11:21:09]--~--~--~--
[11:21:09]FujiNet 0.5.cec57d6a 2021-12-11 17:57:01 Started @ 5
[11:21:09]Starting heap: 4375827
[11:21:09]Detected Hardware Version: 1.6.1 and up
[11:21:09]Enabled Safe Reset Button C
[11:21:09]SPIFFS mounted.
[11:21:09]SD mounted.
[11:21:09]fnConfig::load
[11:21:09]Load fnconfig.ini from SD
[11:21:09]fnConfig::load read 733 bytes from config file
[11:21:09]SPIFFS Config Storage: Enabled
[11:21:09]fnConfig::load read 733 bytes from SPIFFS config file
[11:21:10]WIFI_EVENT_STA_START
[11:21:10]WiFi connect attempt to SSID "CIA-68"
[11:21:10]esp_wifi_connect returned 0
[11:21:10]disk UNMOUNT
[11:21:10]disk MOUNT
[11:21:10]ATR MOUNT
[11:21:10]mounted ATR: paragraphs=5760, sect_size=128, sect_count=720, disk_size=0
[11:21:10]Creating a default printer using FS_SDFAT storage and type 17
[11:21:10]SSH client initialized.
[11:21:10]ModemSniffer::ModemSniffer(FS_SDFAT)
[11:21:10]SIO SETUP
[11:21:10]Set HSIO baud from 0 to 67431 (index 6), alt=68209
[11:21:10]Available heap: 4167131
[11:21:10]Setup complete @ 452 (447ms)
[11:21:10]WIFI_EVENT_STA_CONNECTED
[11:21:11]IP_EVENT_STA_GOT_IP
[11:21:11]Obtained IP address: 192.168.1.71
[11:21:11]SNTP client start
[11:21:11]Starting web server on port 80
[11:21:15]SNTP time sync event: Fri Dec 24, 11:21:15 2021 -0800

tschak909, 2021-12-24:
reproduced. The B button double tap causes a dump of all the running tasks in the debug window. It looks like the data structures for this have changed upstream. Am investigating.

tschak909, 2021-12-24:
fixed.

```
13:53:51.187 > BUTTON_B: DOUBLE-TAP
13:53:51.189 > T01 0x3ffb61b0 (10,10) 3500h   51138792r    Ready: fnLoop
13:53:51.190 > T02 0x3ffbd808 ( 1, 1) 3472h      17968r  Running: fnKeys
13:53:51.191 > T03 0x3ffb33dc ( 0, 0)  976h   50801829r    Ready: IDLE
13:53:51.191 > T04 0x3ffb3b60 ( 0, 0) 1004h     267507r    Ready: IDLE
13:53:51.191 > T05 0x3ffb2c58 ( 1, 1) 1812h     300120r  Blocked: main
13:53:51.191 > T06 0x3ffcd470 (18,18) 2732h      45744r  Blocked: tiT
13:53:51.191 > T07 0x3ffafc4c ( 1, 1)  432h       4300r  Blocked: ipc0
13:53:51.191 > T08 0x3ffbd694 (10,10) 1472h         26r  Blocked: card_detect_int
13:53:51.191 > T09 0x3ffb0314 (24,24)  392h     196001r  Blocked: ipc1
13:53:51.191 > T10 0x3ffdfa08 (10,10) 1476h         24r  Blocked: adamnet_reset_i
13:53:51.191 > T11 0x3ffb8334 ( 5, 5) 7096h        868r  Blocked: httpd
13:53:51.192 > T12 0x3ffce438 (20,20)  964h       3533r  Blocked: sys_evt
13:53:51.192 > T13 0x3ffb94b8 ( 1, 1) 2028h       5392r  Blocked: mdns
13:53:51.192 > T14 0x3ffd0cf0 (23,23) 4428h     398617r  Blocked: wifi
13:53:51.192 > T15 0x3ffb14f8 (22,22) 3300h      57692r Suspened: esp_timer
13:53:51.192 > T16 0x3ffb44e4 ( 1, 1) 1464h         20r  Blocked: Tmr Svc
13:53:51.192 >
13:53:51.192 > CPU MHz: 240
```

push forthcoming in the next few mins.

tschak909, 2021-12-24:
Fixed in commit #5a5d6538d8cf5263d93e4512329d005574b40136

closed 2021-12-24 by tschak909