FujiNetWIFI/fujinet-firmware #481: Replace libssh2 with libssh

tschak909, 2022-01-04:
The code in lib/libssh2 no longer works correctly, due to a combination of used deprecated protocol KEX algorithms, and changes to ESP-IDF, so, it should be replaced with libssh, and connected to the lib/network-protocol/SSH.cpp protocol adapter.

* Fold in libssh: https://github.com/tnn2/esp-idf-libssh and make compile.
* Modify lib/network-protocol/SSH.cpp/h and make it use libssh
* Should be able to connect to an SSH host using netcat, with appropriate login/password credentials. pubkey can be added later.

markjfisher, 2022-11-20:
@tschak909 can this be closed?
I could remove the old libssh2 dir and remove the references everywhere that still uses the old ESP32 client.

tschak909, 2022-11-20:
also yup.

closed 2022-11-20 by tschak909