FujiNetWIFI/fujinet-firmware #482: Documentation (wiki) has a buffer overflow (?) bug for example ApeTime C code

e474, 2022-02-07:
Hi!

   In https://github.com/FujiNetWIFI/fujinet-platformio/wiki/Accessing-the-Real-Time-Clock

   there's a test:

if (time_buf[6] != 0) {

on 

char time_buf[6];

I'm not sure what is being tested for, so can't suggest a fix, but I think the index into time_buf is too high. Also, maybe check  OS.dcb.dstats for a return code as well as (one of) the fields in time_buf?

tschak909, 2022-03-17:
@billkendrick ? ^^

billkendrick, 2022-03-18:
Good point about the buffer not being large enough. I'll update the wiki page. IIRC I based the C code on what Wade did in Action! & BASIC, and I'm still not super comfortable with SIO calls, so I can't speak to the `dstats` suggestion. I'd need to dust off my Atari & FujiNet to experiment a bit, I think. (Maybe hack the FN firmware to fail on purpose.) Or, y'know, if someone else wants to do it. ;)

billkendrick, 2022-08-16:
An attempt was made. Thx!

closed 2022-08-16 by billkendrick