FujiNetWIFI/fujinet-firmware #485: Add TLS socket protocol adapter

cstrotm, 2022-02-22:
A generic TLS socket connection via the N: device handler would be useful to access Gemini server 

https://en.wikipedia.org/wiki/Gemini_(protocol)

I'm working on an Gemini server for the ABBUC webpage contents and would like to create a Atari 8bit Gemini Browser

calaluis, 2026-09-24:
As soon as my pull request 1632 is approved and closed, I'll pick it up. It looks like a good challenge.

RichStephens, 2026-10-16:
@cstrotm Is a TLS socket still something you'd use for the Gemini browser? @calaluis you'd offered to take this; still interested? It's much easier now than in 2021: `fnTcpClientSecure` (from #1468) already does the TLS connect and handshake on ESP and PC, so what's left is a `TLS://` protocol over it, client only, with TCP's non-blocking read/available. Two things to plan for: Gemini capsules mostly use self-signed certs, so it needs a way to skip verification, and each TLS session costs real internal DRAM.

cstrotm, 2026-10-17:
@RichStephens yes, I still find a generic TLS socket useful, not only for Gemini.

Having the ABBUC website information as Gemini content just got easier to implement with the relaunch of the ABBUC website.

RichStephens, 2026-10-17:
@cstrotm Thanks, good to know it's still wanted. @calaluis, still yours if you'd like it; if not, say so and it's open for anyone.

Ghostveri, 2026-11-09:
I'm working on it.