FujiNetWIFI/fujinet-firmware #500: Built-in JSON parsing returns large integers in exponential notation

billkendrick, 2022-05-29:
When parsing http://api.open-notify.org/iss-now.json using #FujiNet's new built-in, SIO-based JSON parsing functionality, the `timestamp` value comes back as a string containing the value in scientific notation / exponential notation / E-notation.  e.g., instead of "`1653818447`", I receive "`1.65382e+09`".

Receiving the large number unaltered, or providing a way for me to query it in such a way that it comes back unaltered (so I can pass it along to `atol()` in my C program on the Atari) would be very helpful.

tschak909, 2022-05-29:
I've changed numerical output to use integers instead of doubles. Let's see if that works.
https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/fnjson/fnjson.cpp#L110

billkendrick, 2022-05-29:
That helped, thanks! :-D

closed 2022-05-29 by billkendrick