FujiNetWIFI/fujinet-firmware #541: Significant Compatibility Issues

miraixyz, 2023-01-01:
Hello.

I am not sure if this has already been brought, but since I don't see any open issues about it, I thought I would post it.

It seems that there is a significant compatibility issue with a lot (a majority) of the Apple II images online.  Only a small handful seem to work.  I know that the files do work because they boot with no issues using with something like Floppy Emu.  

Also, I am using the November 2022 Beta 2 version as the December 2022 version seems to have issues using the ESC key to reboot the Apple.

tschak909, 2023-01-04:
Until @jeffpiep gets the Disk ][ emulation into a usable state, the device will only work with ProDOS images (either .po or .2mg format), as ProDOS has the required block driver.

miraixyz, 2023-01-05:
Well @jeffpiep, let's get it going! I had high expectations for this thing!

closed 2023-07-25 by tschak909

tschak909, 2023-07-25:
> was something done?

Disk ][ support (and support for DO/DSK and WOZ) was added.