FujiNetWIFI/fujinet-firmware #542: Problem accessing APETIME when run along with using APE

XeAtari, 2023-01-04:
Hi,

when FujiNet and APE are used together and APETIME SIO call is performed both devices answer.

I suggest following options:

- config option for APETIME function in FujiNet (power-cycle-proof) that allows for [permanent off | permanent on | auto-soft-disable]
- Like FujiNet keeping quiet if it detects a remote D1: answering upon boot it could soft-disable APETIME in this case

tschak909, 2023-01-04:
Why are you even doing this?

XeAtari, 2023-01-04:
Hi Thom,

thank you very much for your reply.

 > Why are you even doing this?

Why am I doing what?

Are you asking about why I use FujiNet and APE together?

There is an easy answer to this: all my stuff is on a notebook and I run 
a licensed version of APE since, well, maybe two decades or more.

I can hardly do testing and programming for FN with all data residing on 
FN like dismounting the slot where the ATR with my working files.

Or do you know why I run APETIME?

In case you answer why I might use APETIME: well, I always use a disk 
verion of SpartaDOS and in order to have proper time and date stamps for 
the files I need to run APETIME, don't I? I do not want to punch-in date 
and time manually using CLI ...

Kind regards

tschak909, 2023-01-04:
You can't transfer the images to an SD card and put into the FujiNet? There
is no reason to do this, because FujiNet literally provides an APEtime
implementation, and completely supplants all of APE's features.

-Thom

XeAtari, 2023-01-05:
The FujiNet is a really nice device and comes with a ton of features.

But when it comes to _working_ with the Atari, not playing, then APE 
running on the Notebook will always be my main device. There is much 
more easy in searching for stuff and simply mounting using drag'n'drop. 
I can mount a Windows folder as a drive for the Atari, do source code 
editing on the PC and assemble or compile stuff on the A8 using Mac/65 
to a local real disk or any other storage device. Beyond this stuff has 
to be sorted, documentation to be read, data has to be backed-up at 
least every now and then including a backup history and sometimes even 
recovered. Files' data integrity has to be verfified using MD5 checksums.

All this cannot easily be handled if all stuff was on the SD card inside 
the FujiNet module.

My A8 software and dokumentation archive is and will be maintained on 
the notebook. It is the sync source for other mass storage devices like 
SIO2SD, SIO2USB, SDriveMAX, The!Cart and so on.

I guess there will be other people who do not use the FujiNet module as 
their sole device. Maybe just to copy data to it.

I like FujiNet to be able to provide APETIME. It would be lousy if there 
was need for a different time query command because it would require 
doing multiple time queries in a STARTUP.BAT. But if more than one 
device answers there is a problem.

You do not have to care for this issue. It is _my_ problem.

XeAtari

==

tschak909, 2023-01-05:
I really want to say so much here. You're blindly assuming that FujiNet can't take over those duties, and make your Atari not tethered to a notebook, at the same time, and this is, at least, to me, who has spent the last three years making FujiNet able to handle software development workflows, infuriating.

But ok. I give up. You have your workflow, and damned if anyone is going to convince you otherwise. Nevermind that FujiNet has multiple protocol adapters (including TNFS and SMB, to handle network endpoints at a file level.)

I should have just left this thing as a disk sharer, if all anyone was ever going to do is just blindly use this thing without actually digging into it.

-Thom

tschak909, 2023-01-05:
Fine. Will implement the disable.
-Thom

tschak909, 2023-01-05:
have you done any mounting via the web interface?

kenjennings, 2023-01-05:
Your work on Fujinet has been epic and heroic.

Everyone appreciates you.

Except for odd people who want to run two different file servers for the
Atari at the same time.   I also don't see a purpose for APE where Fujinet
exists.

XeAtari, 2023-01-06:
Hi Thom, if I had known beforehand that my post would cause so much 
trouble, I would not have said a word.

I am aware of the fact that FujiNet offers excellent possibilities and 
am in the process of learning at least a little.

Having developed hardware myself in my younger years, I can imagine 
quite well how much work this huge project has demanded and still demands.

Kind regards

==

XeAtari, 2023-01-06:
Erm, yes.

==

tschak909, 2023-01-06:
I have committed code in the master branch to disable the Apetime device on the FujiNet if the following is placed at the bottom of the [ENABLE] section like this:

```
[ENABLE]
enable_device_slot_1=1
enable_device_slot_2=1
enable_device_slot_3=1
enable_device_slot_4=1
enable_device_slot_5=1
enable_device_slot_6=1
enable_device_slot_7=1
enable_device_slot_8=1
enable_apetime=0
```

I will work with @markjfisher to fold in a change to the web UI to enable/disable it, but this should get you going, once we do a build.

---

I apologize profusely for my unbelievably hyperemotional way that I responded to this. 

I can't assume a user's use case, nor can I force them to do anything they do not want to do, and I have to remember that any frustrations really are inconsequential, especially if people are using what is made. 

We'll get a build out, and let you know when you can install it via the flasher.

-Thom

markjfisher, 2023-01-07:
Ignore first 2 commits, my master got out of sync with upstream and I had to rebase locally.

closed 2023-01-07 by tschak909

XeAtari, 2023-01-12:
Hi,

thank you very much for working on this. I will test things as soon as I 
manage to find some time, but currently I am busy repairing 3 Atari 
1050, one of them being utmost trashy and I had to write an article for 
the mag as well.

@Thom
I really do understand your reaction. You put tremendous work into this 
project and my request might have looked to you like I did not honor 
FujiNet the way it deserves. Please be sure that it was not meant this 
way in the least.

XeAtari