FujiNetWIFI/fujinet-firmware #543: Feature Request: Disk Swap button in WebUI

mozzwald, 2023-01-05:
Would be nice to have a disk swap button in the WebUI that does the same as physical disk swap button A.

XeAtari, 2023-01-05:
I am not only looking for swapping D1: against D2:.
I want to be able to swap any Dx: against any Dy:.
Of course this is beyond the capability of the swap button.
A friend of mine and I started writing FDSWAP for the A8 that is 
destined to handle this.
But if the FN firmware provided such a function, a single SIO call could 
do the swap instead of querying all necessary data first (host slot 
numbers, file paths, filenames and read or write flag) and then remount 
stuff in swapped order.
The tool will even allow for one drive to be empty, so we can perform a 
move.
Well, that's the plan. I do hope that we'll be able to really do it.

XeAtari

dillera, 2026-10-03:
This issue has had no activity for about 4 years, and there have been 3094 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-03:
The WebUI has a Swap button now. It runs the same disk rotate as the physical button A.

The comment above also asks for swapping any Dx: with any Dy: in a single call, which the rotate can't do. If that is still wanted, please open a new issue for it.

closed 2026-10-03 by RichStephens