FujiNetWIFI/fujinet-firmware #559: Add file management functions to webUI

tschak909, 2023-02-25:
Add file management functions to the web user interface.

TODO: come back here and flesh out feature set. Anyone who wants features, please comment.
-Thom

HolgerJanz, 2023-03-13:
Would be a great feature!
So I do not have to swap the SD card every time or start an own TNFS server.
AAA+++

dillera, 2026-10-03:
This issue has had no activity for about 4 years, and there have been 2911 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-03:
This is done. #1536 added a file manager to the web UI (`/files`, with a Browse button in settings) for the SD card: list, download, upload (several files at once), rename and delete, and it uses the web UI password if one is set. WebDAV is also available. If there are more file operations you'd want, please open a new issue that says which.

closed 2026-10-03 by RichStephens