FujiNetWIFI/fujinet-firmware #563: SIO command E6 is not acknowledged (Atari 8-Bit)

XeAtari, 2023-03-12:
Hi,

trying to unmount a host slot using command $E6 the FN does not acknowledge the command.

Below please find the logs:

Using 'COM7' as serial port.
Using '460800' as baud rate.
Showing logs:
[08:34:06] xxxxxxxxxx[08:34:06] 
[08:34:06] --~--~--~--
[08:34:06] FujiNet 0.5.13741956 2023-01-30 22:44:10 Started @ 5
[08:34:06] Starting heap: 4365075
[08:34:06] Detected Hardware Version: 1.1-1.5
[08:34:06] Safe Reset Button C: ENABLED
[08:34:06] SPIFFS mounted.
[08:34:06] SD mounted.
[08:34:06] fnConfig::load
[08:34:06] Load fnconfig.ini from SD
[08:34:06] fopen = /fnconfig.ini : ok
[08:34:06] fnConfig::load read 941 bytes from config file
[08:34:07] SPIFFS Config Storage: Enabled
[08:34:07] fnConfig::load read 941 bytes from SPIFFS config file
[08:34:07] disk UNMOUNT
[08:34:07] disk MOUNT
[08:34:07] ATR MOUNT
[08:34:07] mounted ATR: paragraphs=5760, sect_size=128, sect_count=720, disk_size=0
[08:34:07] Creating a default printer using FS_SDFAT storage and type 2
[08:34:07] ModemSniffer::ModemSniffer(FS_SDFAT)
[08:34:07] SIO SETUP
[08:34:07] Set HSIO baud from 0 to 54905 (index 9), alt=55420
[08:34:07] Available heap: 4277235
[08:34:07] Setup complete @ 687 (682ms)
[08:34:07] WiFiManager::start() complete
[08:34:07] WIFI_EVENT_STA_START
[08:34:07] WiFi connect attempt to SSID "HW0"
[08:34:07] esp_wifi_connect returned 0
[08:34:07] WIFI_EVENT_STA_CONNECTED
[08:34:08] IP_EVENT_STA_GOT_IP
[08:34:08] Obtained IP address: 192.168.177.24
[08:34:08] SNTP client start
[08:34:08] Starting web server on port 80
[08:34:10] SNTP time sync event: Sun Mar 12, 08:34:10 2023 +0100
[08:34:20] 
[08:36:10] CF: 70 f9 01 00 6b
[08:36:10] sioFuji::sio_process() called
[08:36:10] ACK!
[08:36:10] Fuji cmd: MOUNT HOST
[08:36:10] ::mount {1} "fujinet.online"
[08:36:10] ::mount_local Attempting mount of "fujinet.online"
[08:36:10] ::mount_tnfs {1:0} "fujinet.online"
[08:36:10] Calling TNFS::begin
[08:36:10] Resolving hostname "fujinet.online"
[08:36:10] Resolved to address 104.248.6.55
[08:36:10] TNFS mount fujinet.online[104.248.6.55]:16384
[08:36:10] TNFS mount successful. session: 0x2233, version: 0x0102, min_retry: 1000ms
[08:36:10] vfs_tnfs_register "/tnfs0x3ffef458" @ 0x3ffef458 = 0 "ESP_OK"
[08:36:10] COMPLETE!
[08:36:38] 
[08:36:38] CF: 70 e6 01 00 58
[08:36:38] sioFuji::sio_process() called
[08:36:38] NAK!
[08:36:38] 
[08:36:38] CF: 70 e6 01 00 58
[08:36:38] sioFuji::sio_process() called
[08:36:38] NAK!
...

tschak909, 2023-03-12:
Hi @XeAtari ! Thank you for finding this. This command wasn't implemented. I am implementing it now.

-Thom

tschak909, 2023-03-12:
@XeAtari this change has been implemented here in firmware:
https://github.com/FujiNetWIFI/fujinet-platformio/commit/6d0c77bd77241af9dcff2788bdbace04b83a7667

XeAtari, 2023-03-14:
Hi tschak909, thanks a lot for digging into this. I'll try to find time for firmware update and testing next weekend.
CU, XeAtari

XeAtari, 2023-03-18:
Hi @tschak909, I just tried to update my FujiNet, but the latest firmware available is dated 2023-02-26 0.5.83bae4a1. Just in case I looked for a Fujinet Flasher update, but it looks like I am running the latest version (1.4.1) Win x64.

tschak909, 2023-03-18:
It has not been pushed to production yet.

tschak909, 2023-03-18:
@mozzwald has uploaded a new build to production.

If it doesn't work, I please advise you to set up an environment so that we can quickly test fixes, without having to spin a production build, each time.

-Thom

XeAtari, 2023-03-19:
Hi @tschak909, thank you very much for the new firmware. Unfortunately the command does not work. It is properly accepted, but the FujiNet device does not seem to really process things. The host slot stays populated in web UI (browser refreshed) and it stays accessible, so one can go through directories, for example.

> advise you to set up an environment so that we can quickly test fixes,

Are sou saying that I should compile firmwares from sources myself? I have absolutely no idea about doing such stuff on a PC. If that would require using an OS other than Windows that would make things even more complicated ...

tschak909, 2023-03-19:
I understand the apprehension. There is a reason I am suggesting this.

The reason I am asking, is because I don't always have an Atari on my bench. I am doing bring-ups for almost half a dozen platforms at the moment, and I can only have one system on my bench at a time.

We have a few people who are doing firmware development and testing who use Windows, so that isn't an issue.

The instructions for setting up an environment, are here:
https://github.com/FujiNetWIFI/fujinet-platformio/wiki/Board-Bring-Up-Software

and I can help you through any issues you encounter.
-Thom

tschak909, 2023-03-19:
...also, can I see what code you are writing to test this?

tschak909, 2023-03-19:
potential issue here: https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/device/sio/disk.cpp#L292

XeAtari, 2023-03-22:
@tschak909 I am shure that you have more than enough work to do. And of course I do understand that it would help if people can test fixes on their own. I just flew over the "Board-Bring-Up-Software" article ... aargh! I'll try to ask someone who is much more capable than I in programming on a PC than me if he thinks that he can do that.

Btw, I am not using any own code to test things. I am performing direct SIO calls using a debugger on a real Atari 8-bit - just to exclude the chance of bugs on my side.

Best, XeAtari

XeAtari, 2023-03-22:
@tschak909 I tried understanding https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/device/sio/disk.cpp#L292

I guess this is not with reference to this issue?

Looks like it is for "[FujiNetWIFI/fujinet-config-tools] Issue #551 is causing problems (Issue #24)" ...

mozzwald, 2023-03-22:
@XeAtari What exactly is the end goal for your use of this command? In a previous comment you said

> The host slot stays populated in web UI (browser refreshed) and it stays accessible, so one can go through directories, for example.

Are you trying to erase/empty/clear the name in the host slot?

XeAtari, 2023-03-23:
@tschak909 I did not "invent" this command. There is a list of device $70 commands here on a list:

https://github.com/FujiNetWIFI/fujinet-platformio/wiki/SIO-Command-%24E6-Umount-Host

Reading its description I thought that "erase/empty/clear" the host slot is exactly what it should do. You can eject a host slot from the web UI and from the build-in config software, can't you?

But maybe I misunderstood the intended use?

mozzwald, 2023-03-23:
> @tschak909 I did not "invent" this command. There is a list of device $70 commands here on a list:
>
> https://github.com/FujiNetWIFI/fujinet-platformio/wiki/SIO-Command-%24E6-Umount-Host
>
> Reading its [...]

Clearly you did not invent the command :laughing: The command was not implemented and I was not clear on what it was supposed to do. I asked @tschak909 what it was supposed to do and then reread what you wrote above which prompted my question. 

You cannot "EJECT" a host slot from the webui or config. You can edit the host from CONFIG and clear it out if you want but not from the webui. Commands already exist to read host slots and write host slots. It's possible to read the list of hosts, modify it (ie, remove/erase/blank a slot) then write it back to fujinet.

One way to interpret the command is to unmount (not delete) the host by first unmounting any disks from that host, then unmounting the VFS (virtual filesystem) on the esp32 to free up memory, but I suppose end users wouldn't care about that.

So, what you want it to do is: unmount any disks associated with the host slot, unmount the host, remove host 
This is possible and will take some time. Until it's done, you can use the READ/WRITE host slot commands recommended above.

XeAtari, 2023-03-23:
Hi @mozzwald Thanks for your reply. 

>You cannot "EJECT" a host slot from the webui or config.

Really? Sorry for being wrong but I do not have a FujiNet at work so I relied on (incorrect) memories.

If I edit a host slot using CONFIG and make it empty I guess related disks remain in the list but become orphaned?

> So, what you want it to do is: unmount any disks associated with the host slot, unmount the host, remove host

That sounds perfectly logic. Related device slots are emptied and are switched offline.

@tschak909 I guess I have found someone who is capable and willing to compile some intermediate (alpha / beta) firmware versions so that those do not have to be made publically available.

mozzwald, 2023-03-29:
@XeAtari What is the end goal for the use of this command? Are you actually just wanting to delete a host slot and leave it empty?

If you are wanting to change a host slot you still need to *read_host_slots()* and then *write_host_slots()* so you could save a step and just use those two commands to get the same result (instead of *delete_host()* then *read_host_slots()* then *write_host_slots()*).

If you REALLY need a *Delete Host* command, it can be done. Would either rename this command or create a new one since *Unmount* is not the same as *Delete/Remove*

XeAtari, 2023-03-31:
Hi,

we should propably ask the person who invented this command and who put 
it in the Atari SIO commands for Device $70 list what that command was 
supposed to do.

 > since /Unmount/ is not the same
 > as /Delete/Remove/

What effect would UNMOUNT cause on a host slot? If it would kill an 
existing TCP/UDP connection - would that be useful for an user or some 
software to do that?

I thought that command was meant to empty a host slot. Like you said 
this can be done by reading the host slots, calculate the position for 
the entry to be deleted, write the necessary data into that buffer 
position and then write stuff back. But this requires quite more 
programming than just sending a single command frame ...

Well, the command is on the list. So it should at least work. It should 
either do what the original author wanted it to do or it should do 
something else, something that makes sense.

Best

==

mozzwald, 2023-03-31:
> It turned out to not be needed, so I will remove the command from the documentation.

You can leave this in the docs since it's implemented and working now since 6c01f21bd5d47ab83bff3f270757802f192c68b2. Disks associated to the chosen host are unmounted first and the host is finally unmounted.

@XeAtari if you wish to have a delete host command added, please open a new issue for feature request, thanks.

closed 2023-03-31 by mozzwald

XeAtari, 2023-04-01:
Thanks a lot to everybody for your support.

Best, XeAtari

==