FujiNetWIFI/fujinet-firmware #580: ADAM CP/M programs not working properly

wmaalouli, 2023-04-04:
CP/M programs have issues on the ADAM Fujinet. They will run, albeit slower than a digital disk drive, but then either crash or not perform disk I/O properly. No issues when running from a standard VDD.

tschak909, 2023-04-04:
Can you provide more information?
-Thom

mrgw454, 2023-04-04:
> CP/M programs have issues on the ADAM Fujinet. They will run, albeit slower than a digital disk drive, but then either crash or not perform disk I/O properly. No issues when running from a standard VDD.

Can you please let us know what specific image files you are using so we can test the same ones?  Is this happening from TNFS or with SD (or both)?  Thanks

wmaalouli, 2023-04-04:
Hi Thom.

I have attached a disk image for the game Ace of Aces I wrote in Turbo Pascal on the Adam under CP/M 2.2. The game runs perfectly fine using a regular VDD, but with the Fujinet, it’s slow, does not load some data from disk, then eventually crashes with a keypress.

To test, boot up the disk and at the A> prompt enter ACES. The game will load. Press any key to bypass the splash screen and select 3 for OLD CAREER. Enter MAALOULI for pilot name. It will not load the stats file for that pilot from disk but rather use the default stats. Press ENTER 2 more times to get to the main game screen which will draw much slower than it should, then hang. Pressing any key crashes it.

There are also a couple of files on the disk called TEST.COM and TEST.PAS. If you try to delete them, it will look like they are deleted, but then if switch to another disk or reboot the disk image, they will reappear.

Hope that helps.

 

Walid

wmaalouli, 2023-04-04:
Hi.

I have attached a disk image for the game Ace of Aces I wrote in Turbo Pascal on the Adam under CP/M 2.2. The game runs perfectly fine using a regular VDD, but with the Fujinet, it’s slow, does not load some data from disk, then eventually crashes with a keypress.

To test, boot up the disk and at the A> prompt enter ACES. The game will load. Press any key to bypass the splash screen and select 3 for OLD CAREER. Enter MAALOULI for pilot name. It will not load the stats file for that pilot from disk but rather use the default stats. Press ENTER 2 more times to get to the main game screen which will draw much slower than it should, then hang. Pressing any key crashes it.

There are also a couple of files on the disk called TEST.COM and TEST.PAS. If you try to delete them, it will look like they are deleted, but then if switch to another disk or reboot the disk image, they will reappear.

Hope that helps.

 

Walid

tschak909, 2023-04-04:
Can you re-send? Nothing was attached.

mrgw454, 2023-04-04:
Hi Walid.  I am not seeing the attachment.  Can you please try it again or send us a link to download it?

I see there are versions on adamarchive.org, but it shows an EOS and CP/M versions in DSK and DDP formats.  

Thanks

wmaalouli, 2023-04-04:
I checked my sent folder and the file was attached. I wonder if it got stripped by your virus checker.

In any case, here it is again.

wmaalouli, 2023-04-04:
I checked my sent folder and file was attached. I wonder if it got stripped by your virus checker. In any case here it is again. You can also find the disk on the Adam Archive site.

wmaalouli, 2023-04-10:
Hi Thom.Any updates on this issue?

tschak909, 2023-04-10:
We've been trying to reproduce the problem, and haven't been able to (on
two different setups). At this point, you'll need to load the FujiNet
flasher, and use the FujiNet with the debug monitor turned on while doing
the same thing, then paste the output to us.

-Thom

wmaalouli, 2023-04-10:
So on your end the game can load the saved pilot file and play a round without crashing? Just making sure.

tschak909, 2023-04-10:
Yes.

tschak909, 2023-04-10:
Can we get a serial monitor dump? That will help zero in on the problem.
-Thom

wmaalouli, 2023-04-10:
I'll try tonight

wmaalouli, 2023-04-18:
Hi Thom.

Sorry it took so long. Finally got a chance to play with it.

Attached is the Fujinet debug output from the time the Adam is powered on till Aces of Aces crashes and returns to the CP/M prompt. Hopefully you’ll be able to figure it out.

Thanks!

 

Walid

wmaalouli, 2023-04-18:
Here you go
[Fujinet debug output.txt](https://github.com/FujiNetWIFI/fujinet-platformio/files/11265742/Fujinet.debug.output.txt)

mrgw454, 2023-04-18:
> Here you go [Fujinet debug output.txt](https://github.com/FujiNetWIFI/fujinet-platformio/files/11265742/Fujinet.debug.output.txt)

Thank you for sending this.  I don't see anything out of the ordinary in the log.  Since it doesn't appear attaching files is working for you with github, is there any way we can get the same copy of the DSK images you are using by sending us a link to a file sharing site?  Perhaps Google Drive?  The images we get from the adamarchive.org site seem to work for us and I'd like to try your specific set of files.  Thanks!

wmaalouli, 2023-04-19:
So I downloaded the Ace of Aces disk from Adam Archive and that one worked just fine, so I suspect that my original image got corrupted in some way. It’s a complex program with dynamic loading of modules from disk, so likely one of the modules was corrupted without affecting the loading of the main program initially.

I guess problem solved for now. Thanks for looking into it.

closed 2023-04-25 by tschak909