FujiNetWIFI/fujinet-firmware #585: Feature request: email printed files

sl0re10, 2023-04-09:
HP printers have been coming with a dedicated email address that will print files emailed to it. Emailing printed files to a address could serve as de facto wireless printing. 

For other printers, such as brother, phone printing is well supported. People can open emails on their phone and print attachments.

fredlcore, 2023-12-19:
Wouldn't this mainly work for plain text files? As soon as you print out graphics, you would only receive gibberish. Or do you mean that the resulting PDF should be e-mailed to you instead?

tschak909, 2023-12-19:
I would assume  the resulting PDFs. The issue is that this feature relies on explicit configurations that don't exist for most people, anymore. This isn't 1999, and most people don't have their own SMTP server.

fredlcore, 2023-12-19:
There are libraries that facilitate sending e-mails from the ESP32, such as this one here:
https://github.com/mobizt/ESP-Mail-Client
So you could even authenticate against a GMail account and send e-mails through their service. Configuration could be done via fnconfig.ini, right?

tschak909, 2023-12-19:
sure.

fredlcore, 2023-12-20:
I could look into that once I have my setup with a generic ESP32-WROVER working in conjunction with Altirra up and running, since I'm planning to get this one running as well, which is probably in the same area of the codebase:
https://www.facebook.com/groups/fujinetusers/permalink/3597409807207336/?mibextid=uJjRxr
But want to make sure that (at least the former) would be a desired feature, because otherwise I'd not invest time into it...

tschak909, 2023-12-20:
Go for it. :)

-Thom

fredlcore, 2023-12-20:
Will do. I ran into my first problem already:
When adding an external library (such as ESP-Mail-Client), the README.md in the /lib folder tells me to add it in that folder and call it like this: `#include <ESP_Mail_Client.h>`. However, when adding this include line for example in printer_emulator.cpp, I always get `fatal error: ESP_Mail_Client.h: No such file or directory`. Also adding it via PlatformIO's library manager leads to the same result. Is this a problem with the library or any other fault by me or did I just put it in the wrong place? Thanks for any pointers...

tschak909, 2023-12-20:
Try using quotes and a relative path, using vs.code's completion to help.

-Thom

fredlcore, 2023-12-20:
Hmm, using quotes and relative path finds the library and tries to compile it but then fails because `#include <Arduino.h>` cannot be found.
I haven't worked on such large distributed projects before, but when I would normally add a library in platformio.ini under `lib_deps`, this library would be installed for this project and I could just include it using `#include <lib.h>`. However, this does not seem to work here, not sure why.
Please don't bother if there is nothing obviously wrong on my side here, I can also just modify that library and adjust all paths to work, but if it should be as the README.md in /libs says, and this is not working for me here, I assume that I'm doing something wrong here...

tschak909, 2023-12-20:
We don't use Ardiuno. We use ESP-IDF, so the library would need to be
ported.
-Thom

fredlcore, 2023-12-20:
Sure, now things make sense ;). I'll see if that is doable...

tschak909, 2023-12-20:
> Last question for today: How is the project's approach towards external
> libraries? Have these been used at all or only the ones that come with the
> ESP-IDF framework? Because if libraries cannot included automatically as
> dependencies and thus have to be ported or otherwise modified, this means
> that any (potentially critical) update will only be possible if someone
> keeps track of any updated library versions and ports them accordingly, and
> this may not always (or at least continuously) be the case. I've been doing
> this for my project, but these were rather uncritical libs. Just want to
> make sure that this won't be an issue here.

We have indeed used external libraries:

* EdURLParser
* libssh2
* samlib

and others.

We have a get it done philosophy.

-Thom

fredlcore, 2023-12-20:
Great, thanks :)

dillera, 2026-10-03:
This issue was opened about 3 years ago, and the code it refers to has changed since.

- `ESP_Mail_Client.h` no longer exists in the codebase
- `Arduino.h` no longer exists in the codebase
- `platformio.ini` no longer exists in the codebase
- `lib.h` no longer exists in the codebase
- `lib_deps` no longer exists in the codebase

The files it names have changed in 12 commit(s) since then.

Closing as no longer applicable to the current firmware. If you can still reproduce this on a current firmware build, please reopen it (or open a new issue) with the firmware version, the platform, and the steps.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera