FujiNetWIFI/fujinet-firmware #586: Propable bug in 'copy file' (Atari SIO command $D8)

XeAtari, 2023-04-09:
Hi,

I stumbled over a condition where there seem to be problems with reading a file from a TNFS host where the FN logs an error but signals successful command execution to the Atari. In fact a file is copied but I am not sure if the contents matches the source file's data.

Below please find the log for your consideration.

[09:32:02] CF: 70 d8 04 01 4e
[09:32:02] sioFuji::sio_process() called
[09:32:02] ACK!
[09:32:02] <-SIO read 256 bytes
[09:32:02] ACK!
[09:32:02] copySpec: tools/mpe.xex|MPE.XEX
[09:32:02] ::mount {3} "fujinet.pl"
[09:32:02] ::mount_local Attempting mount of "fujinet.pl"
[09:32:02] ::mount_tnfs {3:2} "fujinet.pl"
[09:32:02] ::mount_tnfs Currently connected to host "fujinet.pl"
[09:32:02] ::mount {0} "SD"
[09:32:02] ::mount_local Attempting mount of "SD"
[09:32:02] Type is already LOCAL
[09:32:02] fujiHost #3 opening file path "tools/mpe.xex"
[09:32:02] TNFS open file: "/tools/mpe.xex" (0x0001, 0x01b6)
[09:32:02] File opened, handle ID: 0, size: 133120, pos: 0
[09:32:02] fujiHost #0 opening file path "MPE.XEX"
[09:32:02] fopen = MPE.XEX : ok
[09:32:10] Timeout after 2000 milliseconds. Retrying
[09:32:13] Timeout after 2000 milliseconds. Retrying
[09:32:16] Timeout after 2000 milliseconds. Retrying
[09:32:19] Timeout after 2000 milliseconds. Retrying
[09:32:22] Timeout after 2000 milliseconds. Retrying
[09:32:23] Retry attempts failed
[09:32:23] _tnfs_fill_cache received failure condition on TNFS read attempt
[09:32:25] tnfs_read cache fill failed (4294967295) - abortingCOMPLETE!

XeAtari

mozzwald, 2023-04-29:
The above change is untested, but should send an SIO error if the read fails

XeAtari, 2023-04-29:
Thanks a lot for taking care of this issue.

I'll test things as soon as I can apply the new firmware.

Any ideas why reading the file fails? I cannot tell off hand if I tried 
to mount a not-copyable file to a disk slot and boot/load it instead ...

XeAtari

==

mozzwald, 2023-05-01:
@XeAtari my previous attempt to fix the issue was totally off base and has been reverted. Looking deeper, the problem was actually in sio_copy_file() where there was little error checking during the file transfer. I've added error checking during the transfer which should send sio_error() if the file is not successfully copied over and also remove the corrupt copied file from the destination. See commit b3fd4b6e9424a7bc551bd2dcaaf6af2b393186cf for reference.

I'm also in the progress of testing automatic builds with github on my personal fork of the repo. You are welcome to test the fix with FujiNet-Flasher by downloading the firmware file and using the "Custom Firmware File" option in it.

https://github.com/mozzwald/fujinet-platformio/releases/tag/v1.0-pre3

I'm closing the issue. Feel free to reopen if you have anymore problems with it. Thanks for the bug report!

closed 2023-05-01 by mozzwald

XeAtari, 2023-05-03:
Hi mozzwald,

could you please explain what file I need to browse to using the flasher?

I downloaded the fujinet-ATARI-v1.0-pre3.zip and the 
releases-ATARI.json but I guess I have to unpack the archive. 
Unfortunately there is a bunch of files in the archive ...

firmware.bin looks promising, but I do not want to trash my FN by 
guessing wrong.

Btw, I'ld like it if the flasher had a "Clear log button" so one can 
better log specific actions and only have the log snippet for that action.

Best, XeAtari

==

mozzwald, 2023-05-03:
> could you please explain what file I need to browse to using the flasher? I downloaded the fujinet-ATARI-v1.0-pre3.zip and the releases-ATARI.json but I guess I have to unpack the archive. [...]

It's as simple as choosing the zip file. No need to extract it or get any other files. The flasher will unzip the file and properly flash the individual partitions. Some information on the wiki about this https://github.com/FujiNetWIFI/fujinet-platformio/wiki/FujiNet-Flasher#flashing-a-custom-firmware-file

To clear the log window, you can just click the Serial Debug Output button again which should start it over

XeAtari, 2023-05-04:
Hi mosswald,

thanks a lot for your instructions.

Firmwareupdate seems to have worked properly.

Trying to clear the flashers log window by just pushing the "Serial 
Debug Output" button again leads to following result:

Unexpected error: could not open port 'COM7': PermissionError(13, 
'Zugriff verweigert', None, 5)

(Looks like the flasher is trying to open the port while it is still 
using it).

File copy using absolute path worked (I tried it only once this 
evening). But the 532 byte buffer causes so many tiny reads from the 
TNFS, that the procedure took quite long.

File copy using relative path did not work. It looks like the FujiNet 
automatically prepends a slash as a path prefix and hence the file is 
not found because the leading slash means root.

Attached please find two log files in plain text format.

Please let me know if it would be better to open new issues concerning 
the small buffer and the relative path or if those things cannot be done 
anything about.

Best, XeAtari

==

Using 'COM7' as serial port.
Using '460800' as baud rate.
Showing logs:
[20:22:22] 
[20:22:22] CF: 28 fc 68 e6 70
[20:22:22] CHECKSUM_ERROR
[20:22:22] 
[20:22:22] CF: 70 d8 02 01 4c
[20:22:22] sioFuji::sio_process() called
[20:22:22] ACK!
[20:22:22] <-SIO read 256 bytes
[20:22:22] ACK!
[20:22:22] copySpec: ATARI/DOS/Dos_2.5.atr|DOS25.ATR
[20:22:22] ::mount {1} "fujinet.online"
[20:22:22] ::mount_local Attempting mount of "fujinet.online"
[20:22:22] ::mount_tnfs {1:0} "fujinet.online"
[20:22:22] Calling TNFS::begin
[20:22:22] Resolving hostname "fujinet.online"
[20:22:22] Resolved to address 104.248.6.55
[20:22:22] TNFS mount fujinet.online[104.248.6.55]:16384
[20:22:23] TNFS mount successful. session: 0x1dd5, version: 0x0102, min_retry: 1000ms
[20:22:23] vfs_tnfs_register "/tnfs0x3ffee71c" @ 0x3ffee71c = 0 "ESP_OK"
[20:22:23] ::mount {0} "SD"
[20:22:23] ::mount_local Attempting mount of "SD"
[20:22:23] Setting type to LOCAL
[20:22:23] fujiHost #1 opening file path "ATARI/DOS/Dos_2.5.atr"
[20:22:23] TNFS open file: "/ATARI/DOS/Dos_2.5.atr" (0x0001, 0x01b6)
[20:22:23] File opened, handle ID: 0, size: 133136, pos: 0
[20:22:23] fujiHost #0 opening file path "DOS25.ATR"
[20:22:23] fopen = DOS25.ATR : ok
[20:22:23] ::get_filesize
[20:22:23] tnfs_lseek currpos=0, pos=0, typ=1
[20:22:23] tnfs_lseek success, new pos=0, response pos=0
[20:22:23] tnfs_lseek currpos=0, pos=0, typ=2
[20:22:23] tnfs_lseek success, new pos=133136, response pos=133136
[20:22:23] tnfs_lseek currpos=133136, pos=0, typ=0
[20:22:23] tnfs_lseek success, new pos=0, response pos=0
[... 274 lines cut ...]
[20:35:13] TNFS mount successful. session: 0x47a8, version: 0x0102, min_retry: 1000ms
[20:35:13] vfs_tnfs_register "/tnfs0x3ffee7a4" @ 0x3ffee7a4 = 0 "ESP_OK"
[20:35:13] ::mount {0} "SD"
[20:35:13] ::mount_local Attempting mount of "SD"
[20:35:13] Setting type to LOCAL
[20:35:13] fujiHost #1 opening file path "Dos_2.5.atr"
[20:35:13] TNFS open file: "/Dos_2.5.atr" (0x0001, 0x01b6)
[20:35:13] ERROR!
[20:35:13] 
[20:35:13] CF: 70 d8 02 01 4c
[20:35:13] sioFuji::sio_process() called
[20:35:13] ACK!
[20:35:13] <-SIO read 256 bytes
[20:35:13] ACK!
[20:35:13] copySpec: Dos_2.5.atr|DOS25B.ATR
[20:35:13] ::mount {1} "fujinet.online"
[20:35:13] ::mount_local Attempting mount of "fujinet.online"
[20:35:13] ::mount_tnfs {1:2} "fujinet.online"
[20:35:13] ::mount_tnfs Currently connected to host "fujinet.online"
[20:35:13] ::mount {0} "SD"
[20:35:13] ::mount_local Attempting mount of "SD"
[20:35:13] Type is already LOCAL
[20:35:13] fujiHost #1 opening file path "Dos_2.5.atr"
[20:35:13] TNFS open file: "/Dos_2.5.atr" (0x0001, 0x01b6)
[20:35:13] ERROR!

mozzwald, 2023-05-04:
> Trying to clear the flashers log window by just pushing the "Serial Debug Output" button again leads to following result: Unexpected error: could not open port 'COM7': PermissionError(13, 'Zugriff verweigert', None, 5) (Looks like the flasher is trying to open the port while it is still using it).

Ah, ok. For me on Linux clicking the button closes and reopens the port. Apparently this doesn't work the same on Windows. You can open an issue for this in the flasher repo https://github.com/fujiNetWIFI/fujinet-flasher

> File copy using absolute path worked (I tried it only once this evening). But the 532 byte buffer causes so many tiny reads from the TNFS, that the procedure took quite long.

This limit is due to TNFS using UDP packets for transmission and is needed to ensure packets are not dropped. This won't be changed. @tschak909 can clarify if needed

> File copy using relative path did not work. It looks like the FujiNet automatically prepends a slash as a path prefix and hence the file is not found because the leading slash means root.

I believe you need to always send the full path and not sure if there is a way around this. @tschak909 can you clarify here?

XeAtari, 2023-05-04:
Hi,

thanks for that fast answer.

Just for your information (I am not mourning):

I do understand the UDP packet size problem. But this way one will never 
be able to copy a 16 MB file (the Atari can handle disk images of this 
size using old SpartaDOS and I guess even 32 MB using actual SpartaDOS 
X), because the copy procedure (for 16 MB) would take more than 90 
minutes (if I calculated correctly) which is a timeout the Atari SIO 
cannot handle (max is 255 seconds, I guess).

Kind regards, XeAtari

==

tschak909, 2023-05-04:
I understand, which is why the copy command needs to be re-engineered. It was quickly put together, and wasn't meant for large images.

-Thom